espressif/esp-mdf #91: [Question]About the developmen_kit button demo?

Shaopus, 2019-10-29:
Hello,

I found that there is **no low-power function** in the demo, which does not match the ReadMe.

Pls reply soon, Thanks!

zhanzhaocheng, 2019-11-04:
1. All nodules simultaneously perform sleep mode functions, we are currently developing
2. The sleep mode we are using now is: leaf nodes for sleep https://github.com/espressif/esp-mdf/tree/master/examples/development_kit#overview
* [ESP32-MeshKit-Sense](https://github.com/espressif/esp-iot-solution/blob/master/documents/evaluation_boards/ESP32-MeshKit-Sense_guide_en.md): Development board, specifically designed for applications where ESP-MESH is in Light-sleep or Deep-sleep mode. The board provides solutions for:
   * Monitoring the power consumption of MeshKit peripherals
   * Controlling MeshKit peripherals based on the data from multiple onboard sensors. 
  
* [ESP32-MeshKit-Button](button/README.md): Smart button solution, tailored for ESP-MESH applications with ultra-low power consumption. The device wakes up only for a short time when the buttons are pressed and transmits packets to ESP-MESH devices via [ESP-NOW](https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/network/esp_now.html).

Shaopus, 2019-11-04:
> 1. All nodules simultaneously perform sleep mode functions, we are currently developing
> 2. The sleep mode we are using now is: leaf nodes for sleep [...]

Hi, thank you for your reply!
But in the **ESP32-MeshKit-Button** example, in the button.c file, I can't find the application of **deep sleep** in the **button_espnow_mode()** function.
And I could see **esp_sleep_enable_ext1_wakeup()** in the **[ESP32-MeshKit-Sense](https://github.com/espressif/esp-mdf/blob/master/examples/development_kit/sense/main/sense_main.c)**, But in this, I never find it.
Pls reply soon, Thanks!

zhanzhaocheng, 2019-11-04:
ESP32-MeshKit-Button is an ultra-power solution. When it is not working, it disconnects directly from the power supply and therefore does not go into deep sleep.
The power is connected only when the button is triggered and the CPU is woken up to send a packet. After sending the packet, the power will be turned off. When not in use, the ESP32-MeshKit-Button is handled in the off state, so the ESP32-MeshKit-Button can be used for a long time.
https://github.com/espressif/esp-mdf/tree/master/examples/development_kit/button#power-consumption-and-time-characteristics

Shaopus, 2019-11-04:
> ESP32-MeshKit-Button is an ultra-power solution. When it is not working, it disconnects directly from the power supply and therefore does not go into deep sleep.
> The power is connected only when the [...]

Shaopus, 2019-11-04:
Thank you! I know.

closed 2019-11-04 by Shaopus