espressif/esp-mdf #93: mdf upgrade触发 taskWdt

Nepenthes, 2019-11-04:
mesh网络节点74个，向服务器下载固件过程或在向子节点分发固件过程时，主节点重启，打印显示OTA任务触发任务看门狗，该如何解决这个问题呢？ 另外，问下MDF能开启双核吗，为什么MDF menuconfig中没有开启双核的选项

zhanzhaocheng, 2019-11-04:
1. 你可以减少根节点的升级时的日志信息. 这个般为日志太多引起的.     `esp_log_level_set("mupgrade_root", ESP_LOG_INFO)`;
2. MDF 可以开启双核, 如下:
[image]

Nepenthes, 2019-11-04:
还有一个问题，因为产品需求，我将iot-solution中lvgl显示相关的components移植到了MDF中，同时外扩了SPI_RAM，一开始也确实工作起来了，但是spi_lcd驱动部分容易产生奔溃，特别是远程升级的时候，问题打印定位点在esp-iot-solution/components/spi_devices/lcd/spi_lcd.c 第133行lcd数据写入的assert部分，这其中是有什么互斥问题存在吗，还是说mdf的API阻塞很严重，我想问下我应该怎么解决这个问题，或者说使用中我应该注意什么?

zhanzhaocheng, 2019-11-04:
你提供一下完整的日志和 elf 文件

Nepenthes, 2019-11-05:
[dump20191105.zip](https://github.com/espressif/esp-mdf/files/3806703/dump20191105.zip)
dump相关文件已上传，节点越多几率越大，当前dump测试节点数量为45个

zhanzhaocheng, 2019-11-05:
## 原因
这是由于内部的内存不足引起的, spi_master 申请不到内存因此触发了 crash.  由于 OTA 升级时 Wi-Fi 会缓冲部分的数据包.因此在升级时易复现此问题
```shell
I (281008) spi_master: Allocate TX buffer for DMA
assertion "ret == ESP_OK" failed: file "D:/Prj_temp/proj_ESP32/prj_ESP32_reEditByUsr/esp-mdf/components/spi_devices/lcd/spi_lcd.c", line 140, function: lcd_dat
```
## 解决方式
将内部的内存放置在 spi ram 中,如下方式去更改 `make menuconfig`

- 修改 SPI RAM config
[image]

- 修改 BLE 的配置
[image]

Nepenthes, 2019-11-05:
我之前是用的外部内存，按照你要求的配置跑了一次，还是那个地方崩，还有没有别的原因导致呢？
[dump.zip](https://github.com/espressif/esp-mdf/files/3807246/dump.zip)

zhanzhaocheng, 2019-11-06:
从你提供的日志上看,就是内部 RAM 空间不足引发的,你打开一下内存调试功能,我们输出一下内存 RAM 空间看看
```
MDF_LOGI("Free heap, internal current: %d, minimum: %d, total current: %d, minimum: %d",
             heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL),
             heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL),
             esp_get_free_heap_size(), esp_get_minimum_free_heap_size());
```

[image]

Nepenthes, 2019-11-06:
内存调试打开就报告片内内存不足，开启双核时也这样，这是我当前的sdkConfig文件
[image]
[sdkconfig.zip](https://github.com/espressif/esp-mdf/files/3813325/sdkconfig.zip)

zhanzhaocheng, 2019-11-06:
你这个是另一个问题, iram 不足造成的错误,你可以通过, 如下命令是查看RAM 分配情况 :
```
make size-components
```

MGDG, 2019-11-08:
你好，我也遇到iram不足的问题，看了下目前占用89.0%，但是i2c还是经常提示内存开辟失败，不知道有什么解决方案没
```
Total sizes:
 DRAM .data size:   13672 bytes
 DRAM .bss  size:   10096 bytes
Used static DRAM:   23768 bytes ( 100812 available, 19.1% used)
Used static IRAM:  116649 bytes (  14423 available, 89.0% used)
      Flash code:  909699 bytes
    Flash rodata:  366628 bytes
Total image size:~1406648 bytes (.bin may be padded larger)
```

Nepenthes, 2019-11-08:
IDF 指南里面说 DMA malloc不能使用片外ram，我原来占用率98.2%，只剩几百个字节，把menuconfig里的编译方式由debug改为reales，我省出了10k，目前没再出现问题了，你还有14k，iic的DMA应该用不了这么多吧

zhanzhaocheng, 2019-11-08:
@MGDG 你的 iram 不足, 可以修改分配部分 dram 到 iram 中。可以通过如下方式修改
1. soc_memory_layout.c 里加一个 RESERVE REGION，`SOC_RESERVE_MEMORY_REGION(0x3ffffc00, 0x40000000, reverse_for_iram);`
[image]
2. 在`esp32.ld`中修改`iram0_0_seg`的`len`為`0X20400`
`iram0_0_seg (RX) : org = 0x40080000, len = 0x20400`

MGDG, 2019-11-08:
@Nepenthes 我的编译方式已经是reales模式了，并且关闭了其他不需要的功能，用debug模式编译直接报错了。
@zhanzhaocheng 谢谢你的回答，我会按照你的提议修改一下试试