espressif/esp-mdf #94: Increase size of custom data in mconfig through espMesh app

Kasuhikdas, 2019-11-05:
Development Kit: [ESP32-Wrover-Kit]
Core (if using chip or module): ESP-WROOM32
Development Env: [Make]
Operating System: [Ubuntu]
Power Supply: [USB|external 5V]

I want to increase the size of custom data of mconfig though espMesh app.
To do that i have increased the default size 32 to 64 in two files
1.
[image]
2.
[image]

I am able to use 64byte data in the node espMesh send the data first.
But in other nodes i am getting following error.

[image]

How can i fix this issue?

zhanzhaocheng, 2019-11-07:
You need to configure it with `make menuconfig` instead of directly modifying the size of the structure.
[image]

Kasuhikdas, 2019-11-07:
thanks. I tried increasing 32 it works. But when i did 64 i got error

E (867) [mwifi, 457]: <MDF_ERR_INVALID_ARG> !(!MWIFI_ADDR_IS_EMPTY(config->mesh_id))

Kasuhikdas, 2019-11-08:
Is there any way to send certificates though mconfig???

zhanzhaocheng, 2019-11-08:
1. Increase to 64 is working, up to 255 bytes, you need to modify `custom_data` in `mconfig_example.c`,
[image]

2. Certificates are usually transmitted after the distribution network is successful, used for device binding.

Kasuhikdas, 2019-11-08:
Got the custom data thing. 
But i did not get the certificates. I want to modify the meshApp and send certificate to the devices. is there any way to do this?

fanbaoying, 2019-11-08:
[image]

This is  send custom data function

Kasuhikdas, 2019-11-11:
@fanbaoying  I have already sent custom data to my device. I want to send the certificate also while sending mconfig. Mconfig have limit of 255 bytes and certificates will come close to 2kb. 
So,custom data is not an option to send the certificates i guess.

Kasuhikdas, 2019-11-11:
@zhanzhaocheng Is there any way i can generate the certificate in device itself??

zhanzhaocheng, 2019-11-11:
The device side supports a custom data communication process.
Receive data for custom communication processes via MDF_EVENT_MCONFIG_BLUFI_RECV
[image]

Kasuhikdas, 2019-11-12:
nothing received in the event from EspmeshApp while configuring
`      case MDF_EVENT_MCONFIG_BLUFI_RECV:

         mconfig_blufi_data_t *blufi_data = (mconfig_blufi_data_t *)ctx;
         MDF_LOGI("Recv Data: %.*s",blufi_data->size,blufi_data->data);

         break;`
Is there anything else i need to do?

closed 2019-11-18 by Kasuhikdas