espressif/esp-mdf #95: MDF max layer can't exceed 25 nodes

tonyshuang, 2019-11-07:
现需要建立一个很长的一维线性mesh网络。即很多layer，每个layer只有1个节点。
但在menuconfig里，Component config->MDF Mwifi->Capacity config->Max number of allowed layers无法选择超过25个.

我尝试直接修改esp-mdf/components/mwifi/Kconfig文件
确实把range从1~25  变成了1~100。
再次menuconfig，可以超过25了。

但build以后的程序，烧写之后，不停的重启。

请问mdf支不支持这种一维线性的mesh网络结构。如果支持的话，需要怎么配置，才可以超过25个节点。最多支持多少一维线性的节点？

zhanzhaocheng, 2019-11-07:
链状的网络的功能,已经开发完成了,我们内部正在测试中,测试通过后会开放出来
这个方案主要针对的是路灯的场景, 最多可以支持 1024 层

tonyshuang, 2019-11-07:
大概什么时候能放出来。我们目前正在启动一个一维线性的项目，想采用你们的mesh方案

zhanzhaocheng, 2019-11-11:
我们本月会放出来

tonyshuang, 2019-12-09:
一直在关注你们的链状方案，我看现在还未放出，大概什么时候放出？

tonyshuang, 2019-12-31:
> 我们本月会放出来

您好，请问你们的链状方案什么时候放出？多谢

MisterScience5, 2020-03-10:
I am also very interested in this, any updates would be very welcome:)

tonyshuang, 2020-03-12:
> 我们本月会放出来

@zhanzhaocheng     您好，请问大概什么时候链状方案放出? 多谢