espressif/esp-mdf #97: mupgrade and a momentally unreachable node

gdumee, 2019-11-22:
## Environment

- Development Kit:      none
- Core : LOLIN32-LITE
- MDF version : 4d9e0f9a794d63b7a9313f4bd4939e725e80223f
- Development Env:	Eclipse
- Operating System:	Windows
- Power Supply:         	external 5V

## Problem Description
Hello
I am using multiple ESP32 devices in mesh, all with the same firmware. I successfully implemented the mupgrade functions to download a new firmware from HTTP, and upgrade all the nodes on the mesh from the root node.

My question is : how to upgrade a node that is diconnected from the mesh at upgrade time (disconnected or powered off)? In that case the root node doesn't upgrade this node, and when it comes back to the mesh network after the others nodes were upgraded, it is not upgraded by the root node.
Do I need to start again the HTTP update, even if the root node is already up to date, or is it possible for the root node, to send its current firmware to the node ? If yes, does a node can send its firmware to others (example, would it possible that a new node, inserted in a mesh network with a new firmware version, sends its firmware to others nodes) ?

EspHuifeng, 2019-11-25:
It is possible to upgrade specified node by call this function
```
/**
 * @brief  Root sends firmware to other nodes
 *
 * @attention Only called at the root
 *
 * @param  dest_addrs     Destination nodes of mac
 * @param  dest_addrs_num Number of destination nodes
 * @param  result         Must call mupgrade_result_free to free memory
 *
 * @return
 *    - MDF_OK
 *    - MDF_ERR_MUPGRADE_FIRMWARE_NOT_INIT
 *    - MDF_ERR_MUPGRADE_DEVICE_NO_EXIST
 */
mdf_err_t mupgrade_firmware_send(const uint8_t *dest_addrs, size_t dest_addrs_num,
                                 mupgrade_result_t *result);
```
mlink_ota_firmware() show how to use it completely. You can find it in mlink_httpd.c

gdumee, 2019-11-25:
Thanks. I already implemented this function to my code, that's how the root node upgrade the other node's firmware. 

In the example mlink_httpd.c, mupgrade_firmware_send() is done after a HTTP download, a call to mupgrade_firmware_init() and multiple calls to mupgrade_firmware_download(). If I rerun this example, the root node will download again the same firmware it already has.

If I try to initialized mupdare with the correct firmawre size, and then call mupgrade_firmware_send(), it returns an error "download incomplete". I wonder if the root node could initialize mupgrade with its current firmware, and send it to the other nodes.

EspHuifeng, 2019-11-29:
You can call mupgrade_firmware_download_finished() after initialized mupgrade if you don't want to download again.

gdumee, 2020-01-19:
I didn't succeeded to use mupgrade_firmware_download_finished() without downloading again. But I used the node list to upgrade only a subset of my nodes. It could have been better to not downalod again the firmware on the root node, but it's working. Thank you for your help

closed 2020-01-19 by gdumee