espressif/esp-mdf #99: mqtt read buffer is too short

wanglemao, 2019-12-10:
在云智能里面启用了本地时控，点击保存的时候，出现如下错误
[err] iotx_mc_read_packet(575): mqtt read buffer is too short, mqttReadBufLen : 1024, remainDataLen : 8740
我尝试修改了mqtt read buffer的长度，但是发现在 mwifi.c 的1113行的代码存在长度判断，限制在0-8096
请问此问题该如何解决，还是说当前的版本不支持本地时控的命令

wanglemao, 2019-12-10:
我修改了mwifi.c的代码，但是依然报错，报错内容是json解析不全，错误如下：
I (77175) [main, 67]: maliyun_linkkit_propery_set_cb: {"LocalTimer":[{"LightSwitch":1,"Timer":"56 17 * * *","Brightness":2,"Enable":1,"IsValid":1},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable"q???p???lid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer":"","Brightness":0,"Enable":0,"IsValid":0},{"LightSwitch":0,"Timer
W (77299) [main, 74]: <MDF_FAIL> cJSON_Parse

EspHuifeng, 2019-12-11:
如果您用的是 MQTT 传输消息。最好是使用 ESP-IDF-v4.0 级以上版本的 MQTT 库。您有两种方法可以升级，一种是进入 esp-mdf/esp-idf/components/mqtt/esp-mqtt 然后使用 git 切换版本；第二种是直接把 ESP-IDFv4.x 里的 mqtt 复制到您的工程目录下的 components 目录里。参考：[同名组件](https://docs.espressif.com/projects/esp-idf/zh_CN/v3.3/api-guides/build-system.html#id11)

wanglemao, 2019-12-11:
@EspHuifeng 我按照你的方法尝试了，发现问题依然出现，我用的是esp-idf master版本的，而且上面出问题的组件在 esp-mdf/components/mwifi/mwifi.c ，仅仅是替换esp-mqtt应该是没有意义的吧

EspHuifeng, 2019-12-13:
这个目前确实有一个最大限制，在 
`esp-mdf/blob/master/components/maliyun_linkkit/maliyun_linkkit_types_priv.h` 的 29 行

wanglemao, 2019-12-16:
@EspHuifeng 是的，我现在已经限制了时控的大小，这样就不会溢出，不过现在有个问题，子设备怎么加入云智能app中？现在网关是能加入，同时可以控制开关、调光、本地时控等，请问哪里有例子介绍关于子设备的加入方式么？还有每个mesh节点都是网关？还是必须要指定网关，然后其他的是子设备？

lqq96, 2019-12-18:
@wanglemao  
（1）当前的方案是基于 mesh 的方案进行设计，网关对应的是 mesh 中的根节点，同时根节点是由 mesh 网络选择出来的，子设备同样使用云智能的配网方式与网关一致。
（2）当前设备未规定子设备与网关的身份，每个设备都可以做网关与子设备，具体身份的切换是基于 mesh 的投票结果。（一个网络内有且只有一个网关， 剩余的设备默认为子设备）
（3）mesh 网络进行投票组网后， 子节点会以子设备的身份向根节点（网关）请求登录云端。

wanglemao, 2019-12-21:
@lqq96 感谢回答，我下周会进行测试
同时还有一个问题，我们做的是针对海外的市场
我按照 国际站设备开发 的要求：
1. 将 maliyun_linkkit_gateway.c 里面的IOTX_CLOUD_REGION_SHANGHAI 修改为 IOTX_CLOUD_REGION_SINGAPORE
2. 修改 FEATURE_MQTT_DIRECT 为预认证模式， 但是我在代码中，并没有发现这个宏定义，请问这个问题该如何解决？
因为第一次接触这部分开发，所以很多都不太清楚，请帮忙回答下，谢谢。

closed 2020-01-09 by wanglemao