espressif/esp-mdf #101: ESP-IDF 4.0 support

ricksondpenha, 2019-12-12:
hello,

Is there any timeframe on when the esp-idf latest supported by esp-mdf?

EspHuifeng, 2019-12-13:
we are working on it, but we are too busy this days. It will be publish in early 2020.

cgfoed, 2019-12-20:
waiting patiently but hopeful, that it's early in 2020 👍

bbinet, 2020-02-14:
I'm also interested to move to esp-idf v4.0 now that it has been released: https://github.com/espressif/esp-idf/releases/tag/v4.0

@EspHuifeng  Any updated timeframe on when esp-mdf will support esp-idf v4.0?

carlosamtech, 2020-02-18:
Hello any plan to support esp-idf 4.x by esp-mdf

Suraj-singh999898, 2020-03-11:
Does it support with esp-idf v4.0

appventiv, 2020-05-07:
Hello devs, any update on this? can we have a testing branch atleast, our project uses esp-idf 4.x, currently this is a blocker for us, if we could get a early release for mesh support just for testing purposes, would be great.

closed 2020-05-27 by ricksondpenha

cgfoed, 2020-05-27:
Is this resolved?

ricksondpenha, 2020-05-27:
Yes, support for esp-idf master branch, although not recommended for production use cases.. we'll have to check for bugs