espressif/esp-mdf #104: Unable to printf in GPIO driver

JDchew, 2019-12-26:
I wish to print the timestamp from certain place come into this function. So I put a getTick here, but somehow it was not able to printf the value. Is it the driver got something wrong ?

    esp_err_t gpio_set_level(gpio_num_t gpio_num, uint32_t level)
    {
        uint16_t time_send =  xTaskGetTickCount();
	    printf("Gpio Set Lv Time: %d\n",time_send);

        GPIO_CHECK(GPIO_IS_VALID_OUTPUT_GPIO(gpio_num), "GPIO output gpio_num error", 
    ESP_ERR_INVALID_ARG);
        if (level) {
            if (gpio_num < 32) {
                GPIO.out_w1ts = (1 << gpio_num);
            } else {
                GPIO.out1_w1ts.data = (1 << (gpio_num - 32));
            }
        } else {
            if (gpio_num < 32) {
                GPIO.out_w1tc = (1 << gpio_num);
            } else {
                GPIO.out1_w1tc.data = (1 << (gpio_num - 32));
            }
        }
        return ESP_OK;
    }

EspHuifeng, 2019-12-27:
I put the print timestamp code to gpio_set_level function like you did and call it at beginning of  app_main(). I got timestamp strings print. Maybe you can try to  make clean and make again.

JDchew, 2019-12-27:
Oh. I did try to clean before, it also won't help.