espressif/esp-mdf #105: mdf编译时候报错是什么原因 idf和mdf版本不匹配？

chenguang3312, 2019-12-27:
`                               ^
/home/mm/esp/esp-mdf/components/mdebug/mdebug_console.c:78:39: note: each undeclared identifier is reported only once for each function it appears in
     ESP_ERROR_CHECK(uart_param_config(CONFIG_CONSOLE_UART_NUM, &uart_config));
                                       ^~~~~~~~~~~~~~~~~~~~~~~
/home/mm/esp/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
/home/mm/esp/esp-mdf/components/mdebug/mdebug_console.c: In function 'mdebug_console_init':
/home/mm/esp/esp-mdf/components/mdebug/mdebug_console.c:164:23: error: 'CONFIG_CONSOLE_UART_NUM' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
     uart_tx_wait_idle(CONFIG_CONSOLE_UART_NUM);
                       ^~~~~~~~~~~~~~~~~~~~~~~
                       CONFIG_ESP_CONSOLE_UART_NUM
/home/mm/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'mdebug_console.o' failed
make[1]: *** [mdebug_console.o] Error 1
/home/mm/esp/esp-idf/make/project.mk:604: recipe for target 'component-mdebug-build' failed
make: *** [component-mdebug-build] Error 2
`

chenguang3312, 2019-12-27:
mdf是直径下载zip压缩包， idf是git clone的，idf工具路径也没问题

ESP-LBB, 2019-12-30:
当前 MDF 适配的是 IDF 3.2.x / 3.3.x 的版本，你获取的 IDF 可能是 4.x 的版本，请确认下 IDF 版本。

new-soul, 2021-05-04:
请教一下，我有装master最新的idf,然后再安装mdf，这样的话mdf里的idf还是要对应版本？我看master的idf有1.1Gb左右，但是mdf里的idf却几有300多Mb。可以共存不？

zhanzhaocheng, 2021-05-06:
可以的共存，运行 mdf 推荐使用其对应的 idf 版本， esp-idf master 版本是未被验证的，可能会引入新的问题