espressif/esp-mdf #110: How to send a struct user-defined variable over the ESP mesh?

ulahijawi, 2020-01-14:
I am currently testing ESP Mesh example “internal communication”. Kindly advise, what are the possible ways to send a struct type over the ESP Mesh in the esp_mesh_p2p_tx_main function. Clearly, the data.data variable is assigned with tx_buf of type uint8_t, however, how can we send and receive a struct user-defined variable? and what should be changed?

Thank you.

zhanzhaocheng, 2020-01-15:
The mesh does not care about the data type when transmitting. When sending, you only need to cast your structure type to (uint8_t *) when sending.

ulahijawi, 2020-01-16:
Thank you very much. It worked.
Please advise, how can I do the reverse process on the receiving side? I would like to cast the data.data variable to its original struct-type value.

Abdol, 2020-01-27:
A response to this query would be really appreciated @zhanzhaocheng.

zhanzhaocheng, 2020-01-31:
When you receive it, you can force it to your original type，It is recommended that you develop based on ESP-MDF, which can simplify your development
[esp-mdf esp-mesh examples](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi)

Abdol, 2020-02-09:
@zhanzhaocheng thank you for the response. Yes, it worked, though only in simple data types like int, float, char, etc. For more complex datatypes such as strings, maps, and arrays, I think they need to be serialized before transmission.

zhanzhaocheng, 2020-02-10:
@Abdol Data does not need to be serialized. For transmission, what you pass is only a piece of memory space

ulahijawi, 2020-02-10:
@zhanzhaocheng Thanks a lot for your response. Could you please give an example at the sending and receiving parties?
Much appreciated.

zhanzhaocheng, 2020-02-10:
You can see this example: https://github.com/espressif/esp-mdf/tree/master/examples/get-started. It transfers the `char *` type