espressif/esp-mdf #111: ESP MLink with SSL Transport support

githubforsank, 2020-01-15:
Hi people, any plans for support of TLS in mLink. The current source only uses HTTPD functions ?

EspHuifeng, 2020-01-20:
We don't have any plans for support TLS in mLink. if you want to use TLS for root to connect with server, you can implement it in root_read_task() and root_write_task(). Read data from server and send it to expected node in root_read_task(); Read data from node and send to server through TLS in root_write_task.

EspHuifeng, 2020-01-20:
> We don't have any plans for support TLS in mLink. if you want to use TLS for root to connect with server, you can implement it in root_read_task() and root_write_task(). Read data from server and [...]

By the way, you can read mqtt_example in "example/function_demo/mwifi/mqtt_example". It shows how to communicate with sever using MQTT, So it is similar to communicate with server using TCP or TLS.