espressif/esp-mdf #113: Error setup esp32 on Ubuntu 18.04

GianmarcoMicrotech, 2020-01-28:
Hi,

I using esp32-WROVER. I developer using the virtual machine with Ubuntu 18.04, as you have advised me.
I have this problem, i don't execute the hello_world project, i don't understand where i wrong. I have follow this guide :  
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-get-prerequisites 
 
Toolchain path: /home/gianmarco/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2019r2
Compiler version: 8.2.0
Python requirements from /home/gianmarco/esp/esp-idf/requirements.txt are satisfied.
mconf-idf is not required on this platform
mconf-idf is not required on this platform
TERM environment variable is set to "xterm-256color"
MENUCONFIG
Loaded configuration '/home/gianmarco/esp/hello_world/sdkconfig'
No changes to save (for '/home/gianmarco/esp/hello_world/sdkconfig')
Project is not inside a git repository, or git repository has no commits
will not use 'git describe' to determine PROJECT_VER.
App "hello-world" version: 1

 
What i must to do for resolved my problem ?

Best regards,

zhanzhaocheng, 2020-01-31:
What is the version of ESP IDF you are using? Compiler 8.2.0 only supports the version of ESP IDF v4.x

esp-idf v4.x    In the terminal where you are going to use ESP-IDF, run:
```shell
cd ~/esp/esp-idf
./install.sh
. ./export.sh
```

GianmarcoMicrotech, 2020-01-31:
Hi,

I have execute the commands that you have request me, i attached the log.

I have also this problem to developer the my application, with eclipse using the virtual machine with ubuntu 18.04.
 

Can you help me ?

Thanks,

Best regards,
Gianmarco Scarpelli 
 

 
*-*-*-*-*-*-*-*-*-*-*-*-*
Microtech Srl

Via Mameli, 94
53043 Chiusi (Si)
http://www.microtechsrl.net
Tel +39 0578 294621
Fax +39 0578 1900218

************************
Attenzione: Le informazioni contenute in questo messaggio di posta elettronica sono private e confidenziali, riservate solo all'attenzione del destinatario. Qualora doveste ricevere questo messaggio per errore, Vi notifichiamo con la presente che ogni diffusione, riproduzione, distribuzione o utilizzo del presente messaggio è altamente proibita. Siete pregati di informare il mittente con un messaggio di risposta e cancellare immediatamente il presente messaggio ed il suo eventuale contenuto, senza copiarlo o aprirlo.
*****************************************************
This e-mail and any file transmitted with it, which may contain confidential and/or privileged material, is legally privileged and intended solely for the recipient. Access to this e-mail by anyone else, use it for any purpose, store, copy, disclose the contents to any other person totally or partially is strictly prohibited and illegal. If you are not the intended recipient(s), please do not read this e-mail, delete this message and any attachments from your system and contact the sender by reply e-mail or by telephone.
*****************************************************

Data: Thu, 30 Jan 2020 18:57:37 -0800

gianmarco@gianmarco-VirtualBox:~/esp/esp-idf$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20191114 (already installed)
Installing Python environment and packages
Installing Python packages from /home/gianmarco/esp/esp-idf/requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/gianmarco/esp/esp-idf/requirements.txt (line 4)) (44.0.0)
Requirement already satisfied: click>=5.0 in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/gianmarco/esp/esp-idf/requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/gianmarco/esp/esp-idf/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/gianmarco/esp/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/gianmarco/esp/esp-idf/requirements.txt (line 11)) (2.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/gianmarco/esp/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/gianmarco/esp/esp-idf/requirements.txt (line 13)) (0.26)
Requirement already satisfied: ipaddress; python_version < "3" in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/gianmarco/esp/esp-idf/requirements.txt (line 11)) (1.0.23)
Requirement already satisfied: six>=1.4.1 in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/gianmarco/esp/esp-idf/requirements.txt (line 11)) (1.14.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/gianmarco/esp/esp-idf/requirements.txt (line 11)) (1.13.2)
Requirement already satisfied: enum34; python_version < "3" in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/gianmarco/esp/esp-idf/requirements.txt (line 11)) (1.1.6)
Requirement already satisfied: pycparser in /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/gianmarco/esp/esp-idf/requirements.txt (line 11)) (2.19)
All done! You can now run:

  . ./export.sh

gianmarco@gianmarco-VirtualBox:~/esp/esp-idf$ . ./export.sh
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.10.2.
[... cut ...]
  /home/gianmarco/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin
  /home/gianmarco/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
  /home/gianmarco/.espressif/tools/cmake/3.13.4/bin
  /home/gianmarco/.espressif/tools/openocd-esp32/v0.10.0-esp32-20191114/openocd-esp32/bin
  /home/gianmarco/.espressif/tools/ninja/1.9.0/
  /home/gianmarco/.espressif/python_env/idf4.1_py2.7_env/bin
  /home/gianmarco/esp/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

zhanzhaocheng, 2020-02-01:
1. Try to clear the original compilation record

```shell
cd esp-idf/examples/get-started/hello-world
rm -r build sdkconfig*
make -j8
```

2. [Ubuntu virtual machine](http://download.espressif.com/esp_share/env/ubuntu1804.ova) that environment has been set up
```shell
git clone --recursive https://github.com/espressif/esp-mdf.git
export MDF_PATH=~/esp/esp-mdf
cp -r $MDF_PATH/examples/get-started/ .
cd  get-started/
make -j8
```

jimmyw, 2020-03-11:
Switch to using cmake and it will work just fine!