espressif/esp-mdf #116: Memory Leakage issue in MUPGRADE

syed00727, 2020-02-13:
## Environment
- Development Kit:      [ESP32-WROOM] **4MB**
- MDF version: https://github.com/espressif/esp-mdf/commits/master
- Development Env:		[Docker / Make]
- Operating System:		[Ubuntu]
- Power Supply:                       [USB]

## Problem Description

Encountering an issue with the MDF-MUPGRADE component. Have copied the
official ESP-MDF repositories and integrated them into our code. The root node downloads the firmware and flashes it onto itself. When transmitting the firmware, the root node after sending a few packets of the firmware, starts returning an ESP_ERR_MESH_NO_MEMORY error. This error keeps popping up for a while and then after that the MESH breaks down.
We have kept the mesh limited to 1 parent node and 1 child node only. The heap size before the OTA transmission begins is around 19k.

### Expected Behavior

Should be able to download the BIN file from the server and transmit it to the mesh, when all the nodes get the bin, they should flash and reboot (root then childs) 

### Actual Behavior

After broadcasting / sending BIN file first 15 - 16 chunks **ESP_ERR_MESH_NO_MEMORY** Error pop-ups (from the root node to child). 

# ROOT NODE LOG
[image]
[image]

# CHILD NODE LOG
[image]

Eventually available heap size is more then 19K. 

### Code to reproduce this issue
 We're using these esp-mdf methods.
https://github.com/espressif/esp-mdf/blob/master/examples/function_demo/mupgrade/main/mupgrade_example.c 

## Partition Table
# Name,   Type, SubType,  Offset,   Size,  Flags
nvs,      data, nvs,      0x9000,   16k
otadata,  data, ota,      0xd000,   8k
phy_init, data, phy,      0xf000,   4k
ota_0,    app,  ota_0,    0x10000,  1920k
ota_1,    app,  ota_1,    ,         1920k
coredump, data, coredump, ,         64K
reserved, data, 0xfe,     ,         128K

zhanzhaocheng, 2020-02-13:
This is because during the upgrade, the Wi Fi and mesh protocol stacks will cache many packets, resulting in insufficient memory. You can use the following methods:

1. Add the buffer of delay reduction in the upgrade
2. Stop other tasks to free more memory during upgrade
3. Optimize the use of memory, such as

[image]

We will also optimize the memory usage of this upgrade process later

faezs, 2020-02-17:
@zhanzhaocheng thanks for the clarification - we've freed up a bit of the heap by adjusting these. Are there any exposed config options for setting how much caching should happen at the wifi and mesh protocol layers?

zhanzhaocheng, 2020-02-19:
The default number of packets cached by ESP-MESH is 32. You can use 'make menuconfig' or 'esp_mesh_set_xon_qsize()`

[image]
[image]

closed 2020-02-27 by syed00727