espressif/esp-mdf #119: Mesh event data is corrupted if a new event occurs before MDF event loop callback is executed

sinope-db, 2020-03-06:
## Environment

- Development Kit:      none
- Core (if using chip or module): ESP32-Wrover
- MDF version : 3ece464dbac91c0ad516ffb8e59311201bfc8a1c
- Development Env:		VisualGDB
- Operating System:		Windows

## Problem Description

There seem to be a possible race condition where MDF event data will be corrupted if there are more mesh events posted before they are processed from the event loop queue.

When events are posted on the event loop queue, a pointer to a shared variable (s_evet_info) for the event data is posted with the events. The variable will contain the last event data.

### Expected Behavior

Event data is valid for every event (posted with the event)

### Actual Behavior

The event data is valid only if no events are pending in the queue.

### Steps to reproduce

Say we have three mesh events that occurs:
Mesh Event 1 with data A
Mesh Event 2 with data B
Mesh Event 3 with data C

Under most conditions this will happen
1. Mesh callback Event 1 with data A --> posted on loop queue
2. MDF event loop callback for Event 1 with data A 
3. Mesh Event 2 with data B --> posted on loop queue
4. MDF event loop callback for Event 2 with data B
5. Mesh Event 3 with data C --> posted on loop queue
6. MDF event loop callback for Event 3 with data C 

But if event 3 occurs before event 2 is processed from loop event queue
1. Mesh callback Event 1 with data A --> posted on loop queue
2. MDF event loop callback for Event 1 with data A 
3. Mesh Event 2 with data B --> posted on loop queue
4. Mesh Event 3 with data C --> posted on loop queue
5. MDF event loop callback for Event 2 with data C
6. MDF event loop callback for Event 3 with data C

EspHuifeng, 2020-03-10:
Thank for you corrention. We will fix it when we adapt to esp-idf 4.0. We suggest you try not use "ctx" to get information. There are all kinds of API that can get MESH information.

sinope-db, 2020-03-10:
@EspHuifeng My application is using a fixed root node and I'm using MDF_EVENT_MWIFI_ROOT_ADDRESS events to determine the root node presence independently from the toDS state. There does not seem to be API (from IDF or MDF) that provide this information, only API for toDS state.