espressif/esp-mdf #121: Finding issue in seting up or installaing ESP MDF

Gopal123456789, 2020-03-31:
- [ ] I have read the documentation [ESP-MDF Programming Guide](https://docs.espressif.com/projects/esp-mdf/en/latest/) and the issue is not addressed there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Core (if using chip or module): [ESP-WROOM32]
- MDF version :                v1.0-beta1(From "https://github.com/espressif/esp-mdf/releases")
- Operating System:		[Windows 10]
- Power Supply:                [USB]

## Problem Description

Followed Procedure 1:
 From Espressif's get started procedure for ESP IDF installation, I followed the procedure from "https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html".
Required tools are installed using ESP-IDF Tools Installer software.
I was able to configure and flash a "hello world" example using CMD prompt on my ESP32 Dev Board.
Similarly I tried following ESP MDF get started from "https://docs.espressif.com/projects/esp-mdf/en/latest/get-started/index.html".
For setting up tool chain I used the ESP-IDF Tools Installer software, but gave the ESP-MDF location as path. At the end when I open CMD prompt shotcut, I get the error message as the screenshot 1 shows. 
[image]

Please let me know if this procedure is wrong and what could be the correct step here. Is there any other tools installer software fro ESP-MDF as available for the IDF.
With this procedure 1, I am stucked here.

Procedure 2:
I followed another procedure according to "https://docs.espressif.com/projects/esp-idf/en/v3.1.5/get-started/index.html". This process says to use the MSYS2 environment to provide Tool chain. I followed according to it also. 
Gave a path by creating a scripting file named "export_idf_path.sh" according to the "add path according to user profile" procedure. First I did ESP-IDF setup, and was able to configure and flash the "hello world" project on to my board. 
Again for ESP-MDF, made another file named "export_mdf_path.sh" and followed the procedure like IDF setup as mentioned its the same procedure.
I am able to see the configuration page. But while making build and flash I get the error of python as shown in the screenshot 2.
[image]
 
Please guide me through this point.
Thank you.

EspHuifeng, 2020-04-01:
Just run the command prompt in screenshot 2 to intall the python dependencies and try again.

Gopal123456789, 2020-04-01:
Ok. Let me try.

jayrawatrj, 2021-04-27:
Did you find any solutions?? actually I have the same issue and trying different solutions for few days but didn't able to build an esp-mdf program.