espressif/esp-mdf #123: Issue Compiling MDF Get-Started with v4.0 esp-idf

thehellmaker, 2020-04-05:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- Core (if using chip or module): [ESP-WROOM32]
- MDF version (``git rev-parse --short HEAD`` to get the commit id.):  9212c07
- Development Env:		[idf.py]
- Operating System:		[MacOS]
- Power Supply:         [USB]

## Problem Description

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. Setup esp-idf -b 4.0 
2. cd esp-msd/examples/get-started
3. idf.py build

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

## Debug Logs
thehellmaker:get-started thehellmaker$ idf.py flash
Note: You are using Python 3.5.4. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
WARNING: IDF_PATH environment variable is set to /Users/thehellmaker/Documents/atom8/esp/esp-idf but idf.py path indicates IDF directory /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf. Using the environment variable directory, but results may be unexpected...
Checking Python dependencies...
Python requirements from /Users/thehellmaker/Documents/atom8/esp/esp-idf/requirements.txt are satisfied.
Running ninja in directory /Users/thehellmaker/Documents/atom8/esp/esp-mdf/examples/get-started/build
Executing "ninja all"...
[1/369] Performing configure step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-configure 
cd /Users/thehellmaker/Documents/atom8/esp/esp-mdf/examples/get-started/build/bootloader && /usr/local/Cellar/cmake/3.17.0_1/bin/cmake -DSDKCONFIG=/Users/thehellmaker/Documents/atom8/esp/esp-mdf/examples/get-started/sdkconfig -DIDF_PATH=/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf -DSECURE_BOOT_SIGNING_KEY=/Users/thehellmaker/Documents/atom8/esp/esp-mdf/examples/get-started -DEXTRA_COMPONENT_DIRS=/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/bootloader -GNinja /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/bootloader/subproject && /usr/local/Cellar/cmake/3.17.0_1/bin/cmake -E touch /Users/thehellmaker/Documents/atom8/esp/esp-mdf/examples/get-started/build/bootloader-prefix/src/bootloader-stamp/bootloader-configure
CMake Error at /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/tools/cmake/idf_functions.cmake:24 (include):
  include could not find load file:

    components
Call Stack (most recent call first):
  /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/tools/cmake/project.cmake:5 (include)
  CMakeLists.txt:19 (include)

CMake Error at /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/tools/cmake/idf_functions.cmake:43 (idf_set_target):
  Unknown CMake command "idf_set_target".
Call Stack (most recent call first):
  /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/tools/cmake/project.cmake:5 (include)
  CMakeLists.txt:19 (include)

-- Configuring incomplete, errors occurred!
[4/369] Building CXX object esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_pagemanager.cpp.obj
FAILED: esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_pagemanager.cpp.obj 
ccache /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -DESP_PLATFORM -DGCC_NOT_5_2_0=1 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3.1-265-g6c9c253ed\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMDF_VER=\"9212c07\" -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/nvs_flash/include -Iconfig -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/spi_flash/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/mbedtls/port/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/mbedtls/mbedtls/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/include/apps -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/include/apps/sntp -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/lwip/src/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/port/esp32/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/port/esp32/include/arch -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/include_compat -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/vfs/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/esp32/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/driver/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/esp_ringbuf/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/tcpip_adapter/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/esp_event/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/log/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/efuse/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/efuse/esp32/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/newlib/platform_include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/newlib/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/freertos/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/app_trace/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/heap/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/soc/esp32/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/soc/include -mlongcalls   -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu++11 -fno-rtti -fno-exceptions -MD -MT esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_pagemanager.cpp.obj -MF esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_pagemanager.cpp.obj.d -o esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_pagemanager.cpp.obj -c /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/nvs_flash/src/nvs_pagemanager.cpp
In file included from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/ext/string_conversions.h:41,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/bits/basic_string.h:6391,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/string:52,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/stdexcept:39,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/array:39,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/tuple:39,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/bits/unique_ptr.h:37,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/memory:80,
                 from /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/nvs_flash/src/nvs_pagemanager.hpp:17,
                 from /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/nvs_flash/src/nvs_pagemanager.cpp:14:
/Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
[... cut ...]
/Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

thehellmaker, 2020-04-06:
I saw another issue saying https://github.com/espressif/esp-mdf/issues/101

v4.0 is not supported currently it runs on 3.3.1 and esp-idf is inside esp-mdf itself. no need to check it out separately

thehellmaker, 2020-04-06:
Using the esp-idf in this repository or 3.3.1 directly 
1. idf.py.build is still failing during compilation 
2. However building using make is succeeding

`[20/307] Building CXX object esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_storage.cpp.obj
FAILED: esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_storage.cpp.obj 
ccache /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -DESP_PLATFORM -DGCC_NOT_5_2_0=1 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3.1-265-g6c9c253ed\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMDF_VER=\"9212c07\" -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/nvs_flash/include -Iconfig -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/spi_flash/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/mbedtls/port/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/mbedtls/mbedtls/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/include/apps -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/include/apps/sntp -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/lwip/src/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/port/esp32/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/port/esp32/include/arch -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/lwip/include_compat -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/vfs/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/esp32/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/driver/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/esp_ringbuf/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/tcpip_adapter/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/esp_event/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/log/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/efuse/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/efuse/esp32/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/newlib/platform_include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/newlib/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/freertos/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/app_trace/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/heap/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/soc/esp32/include -I/Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/soc/include -mlongcalls   -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu++11 -fno-rtti -fno-exceptions -MD -MT esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_storage.cpp.obj -MF esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_storage.cpp.obj.d -o esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_storage.cpp.obj -c /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/nvs_flash/src/nvs_storage.cpp
In file included from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/ext/string_conversions.h:41,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/bits/basic_string.h:6391,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/string:52,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/stdexcept:39,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/array:39,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/tuple:39,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/bits/unique_ptr.h:37,
                 from /Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/memory:80,
                 from /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/nvs_flash/src/nvs_storage.hpp:17,
                 from /Users/thehellmaker/Documents/atom8/esp/esp-mdf/esp-idf/components/nvs_flash/src/nvs_storage.cpp:14:
/Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
[... cut ...]
/Users/thehellmaker/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
[25/307] Building C object esp-idf/tcpip_adapter/CMakeFiles/idf_component_tcpip_adapter.dir/tcpip_adapter_lwip.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

EspHuifeng, 2020-04-07:
Please use 5.2.0 version toolchain to build it

artafakhari, 2020-06-26:
Is there any roadmap for IDF latest version compatibility with MDF?

This version incompatibility makes some issues in programming environment.