espressif/esp-mdf #126: [mupgrade] Cannot install ota update passing either MWIFI_ADDR_ROOT or the root mac address to mupgrade_firmware_send destination address in mupgrade official example

Ks89, 2020-04-06:
## Environment

- Development Kit:    ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Core (if using chip or module): ESP-WROOM-32D
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 9212c07
- Development Env:	CMake with idf.py
- Operating System:	MacOS Catalina 10.15.4 with all updates
- Power Supply:         USB

## Problem Description

If I pass either `MWIFI_ADDR_ROOT` or the plain mac address of the root node to `mupgrade_firmware_send` function, at the end of the process, it's not able to instal the firmware throwing error `ESP_ERR_MESH_NO_ROUTE_FOUND`.

### Expected Behavior

I should be able to pass the constant MWIFI_ADDR_ANY as in your example, but also MWIFI_ADDR_ROOT or the root mac addsress to install the firmware only on the root node.

### Actual Behavior

I'm using a network with only a single root device and the python server as explained in the README of mupgrade example (to replicate the issue on a public code).
If I use exactly your mupgrade example available in this repo, it works perfectly, but when I change file `mupgrade_example.c` to update only the root node (see above) I receive an error.

### Steps to repropduce
Change file `mupgrade_example.c` from this:
```
// line 111
uint8_t dest_addr[][MWIFI_ADDR_LEN] = {MWIFI_ADDR_ANY};
```
to

```
// line 111
uint8_t dest_addr[][MWIFI_ADDR_LEN] = {MWIFI_ADDR_ROOT};
```

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6824
load:0x40078000,len:12072
load:0x40080400,len:6708
entry 0x40080778
I (71) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v3.3.1-265-g6c9c253ed 2nd stage bootloader
I (39) boot: compile time 10:33:09
I (39) boot: Enabling RNG early entropy source...
I (45) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (98) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (106) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (113) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (121) boot: End of partition table
I (125) boot: No factory image, trying OTA 0
I (130) boot_comm: chip revision: 1, min. application chip revision: 0
[... 121 lines cut ...]
I (24462) esp_image: segment 5: paddr=0x00212cf4 vaddr=0x400862ec size=0x021a0 (  8608)
0x400862ec: esp_rom_spiflash_read_status at /Users/ks89/esp/esp-mdf/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413

I (24470) [mupgrade_example, 187]: The service download firmware is complete, Spend time: 1s
D (24471) [mupgrade_root, 392]: packet_num: 148, total_size: 151232
W (27477) [mwifi, 641]: <ESP_ERR_MESH_NO_ROUTE_FOUND> Node failed to send packets, dest_addr: ff:00:00:01:00:00, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3ffd7dd8, size: 45
W (27485) [mwifi, 767]: <ESP_ERR_MESH_NO_ROUTE_FOUND> Root node failed to send packets, dest_mac: ff:00:00:01:00:00
W (27496) [mwifi, 1215]: Mwifi_transmit_write
W (27501) [mupgrade_root, 232]: Request devices upgrade status
D (30507) [mupgrade_root, 240]: xQueueReceive failed
W (30508) [mwifi, 641]: <ESP_ERR_MESH_NO_ROUTE_FOUND> Node failed to send packets, dest_addr: ff:00:00:01:00:00, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3ffd7dd8, size: 45
W (30519) [mwifi, 767]: <ESP_ERR_MESH_NO_ROUTE_FOUND> Root node failed to send packets, dest_mac: ff:00:00:01:00:00
W (30530) [mwifi, 1215]: Mwifi_transmit_write
W (30535) [mupgrade_root, 232]: Request devices upgrade status
D (33542) [mupgrade_root, 240]: xQueueReceive failed
W (33543) [mwifi, 641]: <ESP_ERR_MESH_NO_ROUTE_FOUND> Node failed to send packets, dest_addr: ff:00:00:01:00:00, flag: 0x22, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3ffd7dd8, size: 45
W (33554) [mwifi, 767]: <ESP_ERR_MESH_NO_ROUTE_FOUND> Root node failed to send packets, dest_mac: ff:00:00:01:00:00
W (33565) [mwifi, 1215]: Mwifi_transmit_write
W (33570) [mupgrade_root, 232]: Request devices upgrade status
D (36577) [mupgrade_root, 240]: xQueueReceive failed
D (36577) [mupgrade_root, 302]: MDF_ERR_MUPGRADE_SEND_PACKET_LOSS
D (36577) [mupgrade_root, 403]: Mupgrade_firmware_send unfinished_num: 1
D (36584) [mupgrade_root, 406]: Count: 0, addr: ff:00:00:01:00:00
(...)
```

EspHuifeng, 2020-04-07:
Mupgrade doesn't support MWIFI_ADDR_ROOT temporarily and we will add this feature in the later version. Currently you can only pass the real MAC address of the root node to upgrade it.

Ks89, 2020-04-07:
> Mupgrade doesn't support MWIFI_ADDR_ROOT temporarily and we will add this feature in the later version. Currently you can only pass the real MAC address of the root node to upgrade it.

Understood, thank you for the answer.

closed 2020-04-07 by Ks89