espressif/esp-mdf #127: stack overflow in mesh-with-aliyun sample

lewisxhe, 2020-04-07:
- [* ] I have read the documentation [ESP-MDF Programming Guide](https://docs.espressif.com/projects/esp-mdf/en/latest/) and the issue is not addressed there.
- [ *] I have updated my MDF branch (master or release) to the latest version and checked that the issue is present there.
- [* ] I have searched the issue tracker for a similar issue and not found a similar issue.

## Problem Description
- Use the default configuration mode, configured as the root node

- When running the esp-mdf / examples / aliyun_linkkit / mesh-with-aliyun sample code, a stack overflow occurs after running for about thirty seconds

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6776
load:0x40078000,len:13232
ho 0 tail 12 room 4
load:0x40080400,len:7172
entry 0x40080778
I (74) boot: Chip Revision: 1
I (74) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v3.3.1-265-g6c9c253ed 2nd stage bootloader
I (41) boot: compile time 14:23:24
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 4MB
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (100) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (108) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (115) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (123) boot: End of partition table
I (127) boot_comm: chip revision: 1, min. application chip revision: 0
I (134) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x44d58 (281944) map
I (243) esp_image: segment 1: paddr=0x00054d80 vaddr=0x3ffb0000 size=0x03f1c ( 16156) load
I (249) esp_image: segment 2: paddr=0x00058ca4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /mnt/d/esp/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1779

I (250) esp_image: segment 3: paddr=0x000590ac vaddr=0x40080400 size=0x06f64 ( 28516) load
I (271) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xe41f0 (934384) map
0x400d0018: _flash_cache_start at ??:?

I (599) esp_image: segment 5: paddr=0x00144210 vaddr=0x40087364 size=0x13b68 ( 80744) load
0x40087364: set_chan_dig_gain at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:2469 (discriminator 1)

I (650) boot: Loaded app from partition at offset 0x10000
I (651) boot: Disabling RNG early entropy source...
I (651) psram: This chip is ESP32-D0WD
E (657) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (663) cpu_start: Failed to init external RAM; continuing without it.
I (670) cpu_start: Pro cpu up.
I (674) cpu_start: Application information:
I (678) cpu_start: Project name:     esp-idf
I (683) cpu_start: App version:      9212c07
I (688) cpu_start: Compile time:     Apr  7 2020 14:23:26
I (694) cpu_start: ELF file SHA256:  ccc78de045a79248...
I (700) cpu_start: ESP-IDF:          v3.3.1-265-g6c9c253ed
I (707) cpu_start: Starting app cpu, entry point is 0x4008151c
0x4008151c: call_start_cpu1 at /mnt/d/esp/esp-mdf/esp-idf/components/esp32/cpu_start.c:268

[... 239 lines cut ...]
I (12370) [aliyu_mesh, 373]: Request ntp success, utc=1586243116000
I (12610) SYS: 2020-04-07T07:05:16Z, free heap: 53472, connect:1, layer: 1, self mac: 80:7d:3a:ca:a6:3c, parent bssid: 78:44:fd:02:fd:f2, node num: 1, parent rssi: -39, login_status:0
I (12610) SYS: root mac: 80:7d:3a:ca:a6:3d, Minimum free heap: 43504
I (12620) [mdf_mem, 199]: Free heap, internal current: 55592, minimum: 43504, total current: 53472, minimum: 43504
***ERROR*** A stack overflow in task Tmr Svc has been detected.
abort() was called at PC 0x4008a7b8 on core 0
0x4008a7b8: vApplicationStackOverflowHook at /mnt/d/esp/esp-mdf/esp-idf/components/esp32/panic.c:715

ELF file SHA256: ccc78de045a79248170aa9d14746c85e0a52f32662e3e400073a8f2e3175475d

Backtrace: 0x4008a55c:0x3ffbf570 0x4008a7a1:0x3ffbf590 0x4008a7b8:0x3ffbf5b0 0x4008c01c:0x3ffbf5d0 0x4008b010:0x3ffbf5f0 0x4008afc2:0x3ffae918
0x4008a55c: invoke_abort at /mnt/d/esp/esp-mdf/esp-idf/components/esp32/panic.c:715

0x4008a7a1: abort at /mnt/d/esp/esp-mdf/esp-idf/components/esp32/panic.c:715

0x4008a7b8: vApplicationStackOverflowHook at /mnt/d/esp/esp-mdf/esp-idf/components/esp32/panic.c:715

0x4008c01c: vTaskSwitchContext at /mnt/d/esp/esp-mdf/esp-idf/components/freertos/tasks.c:3507

0x4008b010: _frxt_dispatch at /mnt/d/esp/esp-mdf/esp-idf/components/freertos/portasm.S:406

0x4008afc2: _frxt_int_exit at /mnt/d/esp/esp-mdf/esp-idf/components/freertos/portasm.S:206

Rebooting...
```

lewisxhe, 2020-04-07:
There are two errors in the [development_kit/light](https://github.com/espressif/esp-mdf/tree/master/examples/development_kit/light) example

[1](https://github.com/espressif/esp-mdf/blob/bdb0ebd63fa90a15ea08ee26d76696960775c92a/examples/development_kit/light/main/aliyun_light_example.c#L221)
[2](https://github.com/espressif/esp-mdf/blob/bdb0ebd63fa90a15ea08ee26d76696960775c92a/examples/development_kit/light/main/component.mk#L12)

In aliyun, if there is no wifi information in the flash, it cannot be allocated from the mesh app, including WeChat applet.

EspHuifeng, 2020-04-07:
Please increase freeRTOS timer task stack size to 2560 or lager. I will put this configuration to the sdkconfig.default in the later version . As for mesh app, we will support it later.

lewisxhe, 2020-04-07:
Thanks, it has been resolved.

closed 2020-04-07 by lewisxhe