espressif/esp-mdf #128: about mwifi group

thuLssss, 2020-04-15:
In esp-mdf/examples/function_demo/mwifi/router, how to set up a group? And what a group means?thank you!

zhanzhaocheng, 2020-04-15:
1. Use [`esp_mesh_set_group_id`](https://github.com/espressif/esp-mdf/blob/master/examples/function_demo/mwifi/router/main/router_example.c#L387) to set the group, and [set the data type to group](https://github.com/espressif/esp-mdf/blob/master/examples/function_demo/mwifi/router/main/router_example.c#L109) when sending, the destination address is group_id, and the device with the same group_id will receive the data.

2. See the meaning of the group: [Device grouping](https://docs.espressif.com/projects/esp-mdf/en/latest/api-guides/mwifi.html#device-grouping)

thuLssss, 2020-04-15:
Ok, thank you!