espressif/esp-mdf #129: AWS_IOT_CONFIG.H no such file or directory

sagar448, 2020-04-15:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM32]
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		[visual studio code] 
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

Everything is working fine except the AWS IoT config file is not being found.
aws_iot_config.h no such file or directory

### Steps to repropduce

Just import the aws_iot_config file

Tried to add it as an external directory, also tried different versions, adding it as a third party, but its still not being found.

EspHuifeng, 2020-04-16:
aws_iot component is supported by esp-idf and you need enable it in `menuconfig->component config-> Amazon Web Services IoT Platform`

NaimSakaamini, 2020-04-16:
> aws_iot component is supported by esp-idf and you need enable it in `menuconfig->component config-> Amazon Web Services IoT Platform`

Hi, I'm having the same issue and I tried to enable aws iot platform but kept getting the same error.

NaimSakaamini, 2020-04-16:
> > aws_iot component is supported by esp-idf and you need enable it in `menuconfig->component config-> Amazon Web Services IoT Platform`
> 
> Hi, I'm having the same issue and I tried to enable aws iot platform but kept getting the same error.

So here's what needs to be done:
beside enabling the Amazon Web Services IoT Platform in menuconfig, you need to add the component name in the CMakeLists.tct file
set(COMPONENT_REQUIRES  aws_iot mcommon mconfig mwifi mlink)

you can close this issue if you wish.

sagar448, 2020-04-16:
Thanks Naim you're amazing! That really helped me.

closed 2020-04-16 by sagar448