espressif/esp-mdf #138: connecting the esp mesh app with any node

ankit-thealchemist, 2020-05-24:
Hello,
I am new to the esp wifi mesh. I want to design the home automation to control the leds and fan, which do not need to be connected to the internet, so no router needed . I was going through the examples and found the no router configuration example. Is it possible to create the mesh network using the esp mesh android app and contols the relays.  Is it possible to connect the app to any node in the network and control any node, like  in the ble mesh.  please guide.

EspHuifeng, 2020-05-25:
First question, if you want to use our esp-mesh app, you can not use no router example. The esp mesh app communicate with root device by http protocol, so the root node need connect to AP and get ip address, the internet connection is not necessary. You can read [Mlink](https://docs.espressif.com/projects/esp-mdf/en/latest/api-guides/mlink.html#mlink) to get more information. Second question, App only connect to root node, and root node forward message to specificated node.

ankit-thealchemist, 2021-01-09:
Closing it!

closed 2021-01-09 by ankit-thealchemist