espressif/esp-mdf #139: WiFi Mode STA OR AP?

sagar448, 2020-05-24:
Hello,

I noticed that in the mesh examples of ESP-IDF, when the wifi was initialized it was not set to station mode. It was left open so it defaults to softAP mode. Where as the examples here in the MDF, it sets every device to Station mode. I feel this may be in correct because in the documentation it states that the intermediate nodes need to be softAP AND station in order to forward packets upstread and downstream. Is this correct?

SO would it be correct to set it as SoftAP AND STA or just to leave it open like done in ESP IDF?

Thanks!

EspHuifeng, 2020-05-25:
The configuration of mesh is execute in mwifi_init() and mwifi_start() and it will not set every device to station mode when mesh network is running.

sagar448, 2020-06-05:
Thats my point, theres an error in the examples. it shouldn't be setting every device to station mode. Station mode is only for leafs as they are at the end of the layer.

closed 2020-06-05 by sagar448