espressif/esp-mdf #141: 新提交的MDF1.0,蓝牙和mesh无法同时启用

fzboffice, 2020-05-28:
I (2570) [mwifi, 292]: esp-mdf version: v1.0-20-gec8f0de-dirty
W (2570) wifi:[beacon]interval:100ms
I (2570) wifi:mode : sta (84:0d:8e:c6:20:a4) + softAP (84:0d:8e:c6:20:a5)
I (2590) wifi:Total power save buffer number: 8
I (2590) wifi:Init max length of beacon: 752/752
I (2590) wifi:Init max length of beacon: 752/752
I (2600) mesh: <nvs>read layer:0, err:0x1102
I (2600) mesh: <nvs>read assoc:0, err:0x1102
I (2600) mesh: [IO]disable self-organizing<reconnect>
W (2610) wifi:[beacon]interval:100ms
I (2610) mesh: [CONFIG]invalid router settings, ssid_len:0, ssid:, bssid:00:00:00:00:00:00
I (3580) wifi:Set ps type: 0

E (3580) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!

abort() was called at PC 0x40197533 on core 0
0x40197533: pm_set_sleep_type at ??:?

Backtrace:0x4009645e:0x3ffd6980 0x4009610d:0x3ffd69a0 0x400993ee:0x3ffd69c0 0x40197533:0x3ffd6a30 0x401813c0:0x3ffd6a50 0x4017ff36:0x3ffd6a70 0x400909d2:0x3ffd6a90 0x400995b9:0x3ffd6ac0
0x4009645e: panic_abort at /mnt/d/esp/esp-mdf/esp-idf/components/esp_system/panic.c:330

0x4009610d: esp_system_abort at /mnt/d/esp/esp-mdf/esp-idf/components/esp_system/system_api.c:68

0x400993ee: abort at /mnt/d/esp/esp-mdf/esp-idf/components/newlib/abort.c:46

0x40197533: pm_set_sleep_type at ??:?

0x401813c0: wifi_set_ps_process at ??:?

0x4017ff36: ieee80211_ioctl_process at ??:?

0x400909d2: ppTask at ??:?

0x400995b9: vPortTaskWrapper at /mnt/d/esp/esp-mdf/esp-idf/components/freertos/xtensa/port.c:143

总体来说启动mesh后mdf主动把pstype置为WIFI_PS_NONE了

EspHuifeng, 2020-06-01:
如果使用蓝牙的话可以把 wifi_init() 里的 esp_wifi_set_ps(WIFI_PS_NONE) 删了

fzboffice, 2020-06-01:
> 如果使用蓝牙的话可以把 wifi_init() 里的 esp_wifi_set_ps(WIFI_PS_NONE) 删了

不是这样的，esp-idf4.0的mesh中会主动把ps设置为0

fzboffice, 2020-06-17:
@EspHuifeng 您好，情况有进展吗

EspHuifeng, 2020-06-29:
我们已经确认这个问题，很快会修复，抱歉让您久等了

fzboffice, 2020-07-16:
> 我们已经确认这个问题，很快会修复，抱歉让您久等了

请问进展如何^^

EspHuifeng, 2020-07-23:
由于我们遇到了一些其它的问题导致我们暂时不能切换 esp-idf 的版本 。你可以手动进入 `esp-mdf/esp-idf/components/esp_wifi/lib`，然后切换 commit id 为 `511dd9e12`，然后再测试一下