espressif/esp-mdf #142: mupgrade on a mesh with different firmwares

NaimSakaamini, 2020-05-28:
## Environment

- Development Kit:      [|ESP32-DevKitC]
- Kit version (for DevKitC): [v3]
- Core (if using chip or module): [ESP32]
- Development Env:		[other]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

I have a mesh with a root, intermediate parents and leaves.
The leaves got a different firmware than the root and the intermediate nodes.
how can I upgrade the firmware for (root and intermediate nodes) separately from the (leaves)?
The example provided assumes that all nodes on the mesh got the same firmware.
Thank you

EspHuifeng, 2020-06-01:
If you want to upgrade specific node, you can change target node list (dest_addr[][]) to what you want to upgrade. See example/function_demo/mupgrade/main/mupgrade_example.c:111

NaimSakaamini, 2020-06-01:
> If you want to upgrade specific node, you can change target node list (dest_addr[][]) to what you want to upgrade. See example/function_demo/mupgrade/main/mupgrade_example.c:111

thanks for the reply.
I want to upgrade all leaves with the same firmware, and then root and all intermediate parents nodes with another firmware.
upgrading one by one doesn't seem like the right solution for a bigger mesh.
any advice?

EspHuifeng, 2020-06-01:
Root doesn't know which one is leaf，maybe you can add heartbeat mechanism then node can tell root which type it is. So root can maintain a table and mark which one is leaf.

NaimSakaamini, 2020-06-01:
> Root doesn't know which one is leaf，maybe you can add heartbeat mechanism then node can tell root which type it is. So root can maintain a table and mark which one is leaf.

any link where I can read about this mechanism and how to implement it?

EspHuifeng, 2020-06-02:
There is no implement about this mechanism in esp-mdf, it must be implemented in application layer.