espressif/esp-mdf #143: The root does not advertise itself to the nodes until it connects to the router config provided

KaranRajPradhan, 2020-05-29:
##Environment

- Development Kit:      [ESP32-DevKitV1]
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //2d715dbfb0146079bb07816a035479e19c0abe49
- Development Env:		[VSCode]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description
The root does not advertise itself to the nodes until it connects to the router provided:
```
mwifi_config_t mesh_config = {
    .channel = CONFIG_MESH_CHANNEL,
    .mesh_id = CONFIG_MESH_ID,
    .router_ssid = DEFAULT_WIFI_SSID,
    .router_password = DEFAULT_WIFI_PASSWORD,
};
```

The DEFAULT_WIFI_SSID and DEFAULT_WIFI_PASSWORD are configurable over Bluetooth. And in a case where incorrect ssid and password are entered, the mesh network is never formed because the root is always trying to connect to a router AP that doesn't exist.

If the mesh_config structure above does not have .router_ssid and .router_password configured then the mesh network forms perfectly:
```
mwifi_config_t mesh_config = {
    .channel = CONFIG_MESH_CHANNEL,
    .mesh_id = CONFIG_MESH_ID,
};
```

### Expected Behavior
The root forms a mesh network if the AP provided is not available.

### Expected Behavior
No mesh network is formed when the wifi ssid and/or password is incorrect.

EspHuifeng, 2020-06-08:
You can set one device as root obviously through mesh_config.mesh_type, and that, all device can forms a mesh network. Here is more information:  [User Designated Root Node](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/mesh.html#user-designated-root-node)

KaranRajPradhan, 2020-08-13:
I'm now disabling the wifi connect if it cant connect to the AP so that the device can advertise itself as the root. Thanks!

closed 2020-08-13 by KaranRajPradhan