espressif/esp-mdf #150: "mqtt_example"  ESP_ERR_MESH_INACTIVE Node failed to send packets

MaBaoGW, 2020-06-12:
Hi

I download the latest code (6/9) and use two esp32 to construct a 2 layer mesh to send mqtt data.

The MQTT sever can get publish messages from both two node. The layer 1 node(root) is fine, but sometimes i can see error message from layer 2 node like this "[mwifi, 672]: <ESP_ERR_MESH_INACTIVE> Node failed to send packets, dest_addr: ff:00:00:01:00:00, flag: 0x28, opt->type: 0x08, opt->len: 13, data->tos: 0, data: 0x3ffcad2c, size: 79"

I can't find any info about the error status <ESP_ERR_MESH_INACTIVE>

Does anyone know how to do about this? thx

[2.txt](https://github.com/espressif/esp-mdf/files/4768694/2.txt)

EspHuifeng, 2020-06-30:
This error was recently added. It can be found in esp_mesh.h. Is this problem certain to happen？

MaBaoGW, 2020-06-30:
yeah i just use example code "mqtt_example"  in two esp32 and log uart output. The 2.txt is the ouput from layer 2 node.

EspHuifeng, 2020-07-01:
Hi, which commit do you use

MaBaoGW, 2020-07-01:
I downloaded the whole esp-mdf source code at 6/9 am10:00. Or you can tell me where can i find the commit version in the code i downloaded.