espressif/esp-mdf #154: 应用环境咨询

wanglemao, 2020-07-09:
我们这有个应用环境，我想采用WiFi mesh的方案，但是感觉不一定能适应，想咨询下
大概是这样：
我们工厂有条流水线，流水线传送带上面存在N个垫板，每个垫板上面放置一个无线采集器和公司的产品，一旦无线采集器发现产品异常，则主动将异常上报的监控电脑；同时监控电脑根据传送带的速度，定时发送命令到指定的无线采集器。
一条线大概有200个无线采集器一起工作，想问下，如果采用WiFi mesh是否能满足要求，其实时性、稳定性又如何
在流水过程中，无线采集器会断电，断电时间不一定，根据流水速度，可长可短，最长大概要几分钟，示意图如下：
[image]

wanglemao, 2020-07-09:
实时性的话，我们要求不高，一般只有产品异常，才会上传数据，监控电脑调光也是间隔几十分钟动作一次

EspHuifeng, 2020-07-20:
Hi，请问你的流水线长度有多长呢？

wanglemao, 2020-07-21:
@EspHuifeng 很近大概20米以内，垫板大概只有20cm，然后垫板挨着垫板，我现在在考虑是否用一个强一点的无线路由器，直接让模块直连路由器

EspHuifeng, 2020-07-21:
那无线采集器上电的时间能有多长呢

wanglemao, 2020-07-21:
@EspHuifeng 滚动速度很慢，大概断电能有几分钟，上电的话，几十分钟吧

EspHuifeng, 2020-07-22:
抱歉，昨天突然有事没能及时回复。还有一个问题，每个采集器多长时间发送一次数据呢，数据长度大约是多少？

wanglemao, 2020-07-22:
@EspHuifeng 几十秒一次吧，我们能接受的最长间隔1分钟，数据内容很短，不会超过100个字节

EspHuifeng, 2020-07-22:
WiFi mesh 应该是可以满足要求的。不过当无线采集器断电的时候，如果这个无线采集器下面还有子节点，那么会导致这些子节点短时间内断开连接，然后重新连接到其他节点，这期间无法发送数据。其他的没有什么问题。