espressif/esp-mdf #155: huge software memory leak in mqtt_example

bonastreyair, 2020-07-10:
## Scenario
In this `switch/case` scenario that gets triggered when MQTT is connected:
https://github.com/espressif/esp-mdf/blob/df0a82554a56a5cb057f009106dfd3d2aee5161c/examples/function_demo/mwifi/mqtt_example/main/mqtt_example.c#L285-L286

It creates the two root communication tasks:
https://github.com/espressif/esp-mdf/blob/df0a82554a56a5cb057f009106dfd3d2aee5161c/examples/function_demo/mwifi/mqtt_example/main/mqtt_example.c#L301-L304

## Issue
The memory leak I found happens when for some reason the root gets bad connection with the broker and gets disconnected and then it connects again. Then we can see that this `case MDF_EVENT_CUSTOM_MQTT_CONNECT` gets triggered again and it creates two more tasks! But previous ones don't get deleted...
...since the root tasks have a delay when MQTT broker is not connected.
https://github.com/espressif/esp-mdf/blob/df0a82554a56a5cb057f009106dfd3d2aee5161c/examples/function_demo/mwifi/mqtt_example/main/mqtt_example.c#L59-L69

## Solution
The creation of those tasks should be in this `case` when the node know it is a root.
https://github.com/espressif/esp-mdf/blob/df0a82554a56a5cb057f009106dfd3d2aee5161c/examples/function_demo/mwifi/mqtt_example/main/mqtt_example.c#L277

I have tested it moving the tasks creation this way and it gets solved as I propose.

EspHuifeng, 2020-07-13:
Yes, you are right, I will fix it as soon as possible. Thank you for your feedback.

bonastreyair, 2020-07-21:
I see this is now solved with the last [commit](https://github.com/espressif/esp-mdf/commit/bdcd590e0bad8cf89684bbcdda12cc91472ef856) right?

EspHuifeng, 2020-07-22:
yes, I fixed this issue based on your solution, thank you for your feedback.

closed 2020-07-22 by bonastreyair