espressif/esp-mdf #159: WiFi Mesh coexist with GATT server and GATT client

owensc84, 2020-07-30:
I would like to use WiFi mesh along side bluetooth low energy.  Are there any examples that show how this is done?  My requirements are to use a GATT server and a GATT client with WiFi mesh.  Any guidance is appreciated.

When I attempt to add a GATT client to the router-less "get-started" project the device is unable to connect to the WiFi mesh network.  The bluetooth does not appear to be advertising either.

I am using esp-mdf at master

This is my app_main.  I tried to highlight where I initialize the GATT client.  If I remove the `init_ancs()` call the WiFi mesh work sas expected.

```c
void app_main()
{

    mwifi_init_config_t cfg = MWIFI_INIT_CONFIG_DEFAULT();
    mwifi_config_t config   = {
        .channel   = CONFIG_MESH_CHANNEL,
        .mesh_id   = CONFIG_MESH_ID,
        .mesh_type = CONFIG_DEVICE_TYPE,
    };

    /**
     * @brief Set the log level for serial port printing.
     */
    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set(TAG, ESP_LOG_DEBUG);

    /**
     * @brief Initialize wifi mesh.
     */
    MDF_ERROR_ASSERT(mdf_event_loop_init(event_loop_cb));
    MDF_ERROR_ASSERT(wifi_init());
    MDF_ERROR_ASSERT(mwifi_init(&cfg));
    MDF_ERROR_ASSERT(mwifi_set_config(&config));
    MDF_ERROR_ASSERT(mwifi_start());

////////////////////////////////////////
    init_ancs();
////////////////////////////////////////    
/**
     * @brief Data transfer between wifi mesh devices
     */
    if (config.mesh_type == MESH_ROOT) {
        xTaskCreate(root_task, "root_task", 4 * 1024,
                    NULL, CONFIG_MDF_TASK_DEFAULT_PRIOTY, NULL);
    } else {
        xTaskCreate(node_write_task, "node_write_task", 4 * 1024,
                    NULL, CONFIG_MDF_TASK_DEFAULT_PRIOTY, NULL);
        xTaskCreate(node_read_task, "node_read_task", 4 * 1024,
                    NULL, CONFIG_MDF_TASK_DEFAULT_PRIOTY, NULL);
    }

    TimerHandle_t timer = xTimerCreate("print_system_info", 10000 / portTICK_RATE_MS,
                                       true, NULL, print_system_info_timercb);
    xTimerStart(timer, 0);

}
```

To follow is the GATT client initialization.  The GATT client comes from the ESP32 ANCS example project.

```c
void init_ancs(void)
//void app_main(void)
{
    esp_err_t ret;

    // Initialize NVS.
    #if 0
    ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
[... 359 lines cut ...]
I (18755) mesh: [FAIL][56]root:0, fail:56, normal:0

I (19062) mesh: [SCAN][ch:13]AP:1, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (19064) mesh: [FAIL][57]root:0, fail:57, normal:0

W (19370) wifi:scan number is 0
I (19370) mesh: [SCAN][ch:13]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (19374) mesh: [FAIL][58]root:0, fail:58, normal:0

W (19681) wifi:scan number is 0
I (19682) mesh: [SCAN][ch:13]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (19686) mesh: [FAIL][59]root:0, fail:59, normal:0

I (19993) mesh: [SCAN][ch:13]AP:1, other(ID:0, RD:0), MAP:1, idle:1, candidate:0, root:0, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (19995) mesh: [FAIL][60]root:0, fail:60, normal:0

I (20000) mesh: <MESH_NWK_PARENT_SELECTION>no parent found, set ROOTLESS, scan times:60, heap:175032
I (20009) mesh: <MESH_NWK_PARENT_SELECTION>channel<cfg:13> is allowed to switch, do a full channel scan
I (20010) [get_started, 187]: event_loop_cb, event: 9
I (20018) wifi:mode : sta (fc:f5:c4:3c:53:d0)
I (20029) [mwifi, 264]: Network state: root_less
I (20033) [get_started, 187]: event_loop_cb, event: 11
I (20034) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x2, need_scan_router:0x0, look_for_nwk_count:0
I (20039) [get_started, 187]: event_loop_cb, event: 20
I (21484) [get_started, 135]: System information, channel: 5, layer: -1, self mac: fc:f5:c4:3c:53:d0, parent bssid: 00:00:00:00:00:00, parent rssi: -120, node num: 0, free heap: 175772

owensc84, 2020-07-30:
I see the error "I (1708) coexist: [854006] Error! Should enable WiFi modem sleep when both WiFi and Bluetooth ar"

I have set the modem sleep, see image below.  Is there another place I need to modify the modem sleep options?
[image]

owensc84, 2020-07-30:
My sdkconfig

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
# end of Bootloader config

#
# Security features
#
# CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT is not set
# CONFIG_SECURE_BOOT is not set
[... 1643 lines cut ...]
CONFIG_TCP_RECVMBOX_SIZE=6
CONFIG_TCP_QUEUE_OOSEQ=y
# CONFIG_ESP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES is not set
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options

owensc84, 2020-07-30:
I turned on wifi modem sleep by, `MDF_ERROR_ASSERT(esp_wifi_set_ps(WIFI_PS_MIN_MODEM));`.  I also moved the bluetooth initialization to before the wifi.  I still crashes the ESP32 with an error about the modem sleep but when it boots back up the error is different -- "invalid router settings."

Logs
```
I (31) boot: ESP-IDF v4.3-dev-134-g625bd5eb1 2nd stage bootloader
I (31) boot: compile time 17:32:16
I (31) boot: chip revision: 3
I (35) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (87) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (102) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (110) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (117) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (125) boot: End of partition table
I (129) boot_comm: chip revision: 3, min. application chip revision: 0
I (136) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3a5b4 (239028) map
I (237) esp_image: segment 1: paddr=0x0004a5dc vaddr=0x3ffbdb60 size=0x04ed8 ( 20184) load
I (245) esp_image: segment 2: paddr=0x0004f4bc vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/eds/call_light/git/esp_mesh/esp-mdf/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (246) esp_image: segment 3: paddr=0x0004f8c8 vaddr=0x40080404 size=0x00750 (  1872) load
I (253) esp_image: segment 4: paddr=0x00050020 vaddr=0x400d0020 size=0x1013c0 (1053632) map
0x400d0020: _stext at ??:?

I (662) esp_image: segment 5: paddr=0x001513e8 vaddr=0x40080b54 size=0x1bd7c (114044) load
I (727) boot: Loaded app from partition at offset 0x10000
I (728) boot: Disabling RNG early entropy source...
I (728) cpu_start: Pro cpu up.
I (732) cpu_start: Application information:
I (736) cpu_start: Project name:     get_started
I (742) cpu_start: App version:      1
I (746) cpu_start: Compile time:     Jul 30 2020 18:31:54
I (752) cpu_start: ELF file SHA256:  d1019f1c637662aa...
I (758) cpu_start: ESP-IDF:          v4.3-dev-134-g625bd5eb1
I (765) cpu_start: Single core mode
I (769) heap_init: Initializing. RAM available for dynamic allocation:
I (776) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (782) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (788) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (794) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (801) heap_init: At 3FFD0FA8 len 0000F058 (60 KiB): DRAM
I (807) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (813) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (819) heap_init: At 4009C8D0 len 00003730 (13 KiB): IRAM
I (825) cpu_start: Pro cpu start user code
I (843) spi_flash: detected chip: gd
I (844) spi_flash: flash io: dio
I (844) esp_core_dump_flash: Init core dump to flash
I (847) esp_core_dump_flash: Found partition 'coredump' @ 3d0000 65536 bytes
I (861) esp_core_dump_flash: Found core dump 29104 bytes in flash @ 0x3d0000
I (862) cpu_start: Starting scheduler on PRO CPU.
I (885) BTDM_INIT: BT controller compile version [826d7c6]
I (887) system_api: Base MAC address is not set
I (887) system_api: read default base MAC address from EFUSE
I (986) phy: phy_version: 4182, f1ba940, Jun  4 2020, 19:40:07, 0, 0
I (1221) BLE_ANCS: init_ancs init bluetooth
I (1349) BLE_ANCS: REG_EVT
I (1384) wifi:wifi driver task: 3ffd85b0, prio:23, stack:6656, core=0
I (1401) wifi:wifi firmware version: 0507408
I (1402) wifi:wifi certification version: v7.0
I (1402) wifi:config NVS flash: enabled
I (1402) wifi:config nano formating: disabled
[... 151 lines cut ...]
I (1481) mesh: [CONFIG]invalid router settings, ssid_len:0, ssid:, bssid:00:00:00:00:00:00
I (2196) wifi:Set ps type: 0

E (2197) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!

abort() was called at PC 0x4019f607 on core 0
0x4019f607: pm_set_sleep_type at ??:?

Backtrace:0x4008d31f:0x3ffd83e0 0x4008d981:0x3ffd8400 0x4009460e:0x3ffd8420 0x4019f607:0x3ffd8490 0x4018ae24:0x3ffd84b0 0x4018994e:0x3ffd84d0 0x40096346:0x3ffd84f0 0x4008d989:0x3ffd8520
0x4008d31f: panic_abort at C:/eds/call_light/git/esp_mesh/esp-mdf/esp-idf/components/esp_system/panic.c:330

0x4008d981: esp_system_abort at C:/eds/call_light/git/esp_mesh/esp-mdf/esp-idf/components/esp_system/system_api.c:100

0x4009460e: abort at C:/eds/call_light/git/esp_mesh/esp-mdf/esp-idf/components/newlib/abort.c:46

0x4019f607: pm_set_sleep_type at ??:?

0x4018ae24: wifi_set_ps_process at ??:?

0x4018994e: ieee80211_ioctl_process at ??:?

0x40096346: ppTask at ??:?

0x4008d989: vPortTaskWrapper at C:/eds/call_light/git/esp_mesh/esp-mdf/esp-idf/components/freertos/xtensa/port.c:143
```

owensc84, 2020-07-30:
My entire project for reference.
[get-started.zip](https://github.com/espressif/esp-mdf/files/5004259/get-started.zip)

EspHuifeng, 2020-07-31:
can you delete this line in you code and try again.
https://github.com/espressif/esp-mdf/blob/3894d4762cbd0682b042e8681f930f5a1d939505/examples/get-started/main/get_started.c#L170

owensc84, 2020-07-31:
> 
> 
> can you delete this line in your code and try again.
> 
> https://github.com/espressif/esp-mdf/blob/3894d4762cbd0682b042e8681f930f5a1d939505/examples/get-started/main/get_started.c#L170

@EspHuifeng , thank you for the suggestion.

I have tried to remove that line and update that line with `MDF_ERROR_ASSERT(esp_wifi_set_ps(WIFI_PS_MIN_MODEM));` with the same result.

owensc84, 2020-08-01:
I took a closer look at the logs, and it looks like wifi sets the power save to 1 (min. modem) and shortly after sets it back to 0 (none).  See logs below.  Any ideas on how to prevent the wifi power save mode from changing to none?

```
I (1030) phy: phy_version: 4182, f1ba940, Jun  4 2020, 19:40:07, 0, 0
I (1032) wifi:mode : sta (fc:f5:c4:3c:55:00)
///////////////////////////////////////////////////////////
I (1033) wifi:Set ps type: 1
///////////////////////////////////////////////////////////

I (1034) [mwifi, 292]: esp-mdf version: v1.0-69-gf8e8807
I (1039) wifi:mode : sta (fc:f5:c4:3c:55:00) + softAP (fc:f5:c4:3c:55:01)
I (1047) wifi:Total power save buffer number: 16
I (1050) wifi:Init max length of beacon: 752/752
I (1054) wifi:Init max length of beacon: 752/752
I (1059) mesh: <nvs>read layer:0, err:0x1102
I (1062) mesh: <nvs>read assoc:0, err:0x1102
I (1067) mesh: [IO]disable self-organizing<reconnect>
I (1072) mesh: [CONFIG]invalid router settings, ssid_len:0, ssid:, bssid:00:00:00:00:00:00
////////////////////////////////////////////////////////////////
I (1683) wifi:Set ps type: 0
////////////////////////////////////////////////////////////////

```

EspHuifeng, 2020-08-03:
Can you checkout esp-idf/components/esp_wifi/lib to 511dd9e12 and try again.

owensc84, 2020-08-03:
> 
> 
> Can you checkout esp-idf/components/esp_wifi/lib to 511dd9e12 and try again.

I have started using the esp-mdf at tag release/v1.0 and the issue cleared itself up.  I will continue to use release/v1.0 in my development and let you know if I have any other issues.

Thanks again for your help!

KaranRajPradhan, 2020-08-19:
I'm facing the same issue however. Using branch release/v1.0 with commit hash gd0196da. Using the following components:
Bluetooth: BLE GATT server
WiFi
aws_iot
Mesh (esp-mdf)

Getting the following crash logs:

```
I (15555) BLE_COMMON: Now initialising BLE
I (15555) BTDM_INIT: BT controller compile version [66c0831]
I (15555) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (15565) coexist: [15540003] Error! Should enable WiFi modem sleep when both WiFi and Bluetooth 
abort() was called at PC 0x400f83f6 on core 0
0x400f83f6: coex_init at ??:?

ELF file SHA256: d3e29ff9752c314f

Backtrace: 0x4008cf5f:0x3fff27f0 0x4008d0e5:0x3fff2810 0x400f83f6:0x3fff2830 0x400d162f:0x3fff2850 0x400d1671:0x3fff2870 0x400d18a5:0x3fff28a0 0x40140448:0x3fff28d0 0x4013e259:0x3fff28f0 0x401925c2:0x3fff2940
0x4008cf5f: invoke_abort at /home/karan/esp/esp-mdf/esp-idf/components/esp32/panic.c:136

0x4008d0e5: abort at /home/karan/esp/esp-mdf/esp-idf/components/esp32/panic.c:171

0x400f83f6: coex_init at ??:?

0x400d162f: esp_phy_rf_init at /home/karan/esp/esp-mdf/esp-idf/components/esp32/phy_init.c:570

0x400d1671: esp_phy_rf_init at /home/karan/esp/esp-mdf/esp-idf/components/esp32/phy_init.c:570

0x400d18a5: esp_phy_load_cal_and_init at /home/karan/esp/esp-mdf/esp-idf/components/esp32/phy_init.c:716

0x40140448: esp_bt_controller_enable at /home/karan/esp/esp-mdf/esp-idf/components/bt/bt.c:1415

0x4013e259: init_ble at /home/karan/code/iotready/joovv/joovv-mesh/firmware/source/components/ble/ble_common.c:113

0x401925c2: aws_iot_task at /home/karan/code/iotready/joovv/joovv-mesh/firmware/source/components/aws/aws_control.c:455

Rebooting...
```

> Can you checkout esp-idf/components/esp_wifi/lib to 511dd9e12 and try again.
@EspHuifeng There seems to be no esp_wifi directory in esp-idf/components/
[image]

EspHuifeng, 2020-08-26:
It would be ok if you init ble before start mesh. But we don't recommend open ble and wifi-mesh at the same time, because wifi-mesh and ble will effect each other.

owensc84, 2020-08-26:
> 
> 
> It would be ok if you init ble before start mesh. But we don't recommend open ble and wifi-mesh at the same time, because wifi-mesh and ble will effect each other.

Why do you say this?  We  are currently developing a product that will utilize BLE and wifi-mesh at the same time.  The esp-mdf documentation states that "Simultaneously run Wi-Fi and BLE protocol stacks."  [Source](https://github.com/espressif/esp-mdf#esp-wifi-mesh-highlights) of that quote.

Are you saying we should go a different direction if we need to access a mesh network an BLE?

EspHuifeng, 2020-09-09:
We recommend not to use WiFi-MESH and BLE at the same time, these two do not currently work well.

hilgertd, 2020-09-09:
@EspHuifeng please explain what you mean? Are you saying BLE and WiFi-Mesh can never be used at the same time?

EspHuifeng, 2020-09-10:
If wifi-mesh and ble work together, the rate and stability of wifi-mesh and ble will be affected both. The current idf ble and wifi libraries will require that WiFi modem sleep (PS type is not 0) mode must be enabled when coexisting, and this parameter will be checked during initialization. due to the wifi-mesh will set PS type to 0 after start. If you don't care about these effects, you can bypass this check, that is, first enable the wifi modem sleep mode, then set the ble, and finally start the mesh.

bbinet, 2020-09-10:
@EspHuifeng Is this compatibility issue planned to be fixed in the (near) future? so that we can use BLE to manage the wifi mesh locally with no internet connection required?

ClaesIvarsson, 2020-10-14:
I would also like to know when this is planned to be fixed. In my project we plan to use BLE GATT server together with mesh wifi.

browningweb, 2020-10-30:
so, any progress? We also plan to use ble together with mesh. ble & mesh wifi gateway.

ABIDI-Abdelfatteh, 2021-11-07:
I have faced the same problem with IDF WIFI-MESH component. Even with enabling power-saving on both WiFi and Gatt a system panic will occur. I fixed this by enabling the WiFi-MESH power-saving mode [esp_mesh_enable_ps()]. Since WiFi-MESH uses WiFi internally it will affect the WiFi settings.
So, try to find-out how to enable mesh power-saving mode on MDF.

benhugsmart, 2022-06-17:
I'm also wondering if BLE + Mesh stability is likely to be improved in the near future? Our application requires both to be used at the same time.

Leadrive, 2023-04-21:
@EspHuifeng 我测试过 esp-mesh-lite项目，启用wifi后，照样是可以用ble的。
麻烦你们更新一下吧，我都为你们感到脸红，那么多人用来做项目，你们说停就停了，有问题也不解决。
关键是这部分还不开源，别人想改还改不了。