espressif/esp-mdf #160: IRAM issue when enable the external RAM

NOA-Vernon, 2020-08-03:
I have a requirement that BLE + wifi mesh + PSRAM should coexist at the same time, is it possible to do that?
After combining the examples together, I found the combined example can't pass the compiling, the errors said "IRAM0 segment data does not fit" and "region `iram0_0_seg' overflowed by 14456 bytes".

Then I comment the BLE function in the combined example and re-compile, it passed and could run normally without BLE feature, but this is NOT what I want.  
In order to save more space of IRAM, I try to modify the compiler options to optimize for size(-Os), however, it can save a little IRAM space but still not enough when combined the BLE + wifi mesh + PSRAM together.
Therefore, I also disable "WiFi IRAM speed optimization" and "WiFi RX IRAM speed optimization" in the menuconfig-->Component config-->Wi-Fi, it can pass compiling, but a BIG problem is that it's failed to start the wifi mesh when BLE had already run, it means the BLE can't coexist with the wifi mesh.

I have run the "idf.py size-components" for checking the IRAM summary, according to the idf.py size-components the libc.a and libesp32.a need much more space in IRAM when the external PSRAM is enabled. (https://esp32.com/viewtopic.php?t=15595) 
Base on the problem, I assume the problem is caused by the iram0_0_seg length defined in the /esp-idf/components/esp32/ld/esp32.ld, only 0x20000(128Kbyte).  I wonder what result would cause if I extend the length of iram0_0_seg, how the IRAM and DRAM formed in the internal SRAM(520KiB), can I use more space of IRAM and less DRAM, or the user application only use the external RAM instead of the internal SRAM?

Thanks.

EspHuifeng, 2020-08-03:
IRAM space is limited and can not contain both ble and wifi, so it is right to disable wifi IRAM speed optimization and WiFi RX IRAM speed optimization. 
The coexist issue is another issue and you can checkout  commit id of esp-idf/components/esp_wifi/lib to 511dd9e12 temporarily and try again.

NOA-Vernon, 2020-08-03:
Thanks a lot for your answer, I tried again and test, it works.

EspHuifeng, 2020-08-03:
You can refer to section 2.3 of the technical manual。 https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

NOA-Vernon, 2020-08-04:
Thank you so much for your answer. @EspHuifeng

closed 2020-08-04 by NOA-Vernon