espressif/esp-mdf #164: error when set  "esp_mesh_set_ap_assoc_expire(int seconds)" under 10 seconds

lybaohuy, 2020-08-17:
this function mean that when it's child is inactive during x seconds, mesh softAP associate will disassociate it.
for my perspective, 10s or higher is quite long, because softAP  will be stuck until timeout( x seconds), and the softAP  will can't send anything later, this mean that our mesh network( maybe partial area or whole) will be stopped in x seconds.

it is better if can change under 10 seconds.

EspHuifeng, 2020-08-26:
The default value of this parameter is 30000 ms in MDF, have you encounter some issue because of this parameter?

lybaohuy, 2020-08-26:
yes, default is 30000ms.
but when i changed it to under 10000ms by esp_mesh_set_ap_assoc_expire(), it always be errorred, must be >=10000ms.

you can try that.

EspHuifeng, 2020-09-09:
Yes, the minimum of this parameter is 10000 ms.