espressif/esp-mdf #165: UART console when change baudrate

lybaohuy, 2020-08-17:
i changed baudrate of UART0 console to 921600 but it can'nt run correctly on the latest master( 17 Aug 2020)
in branch v1.0, it work well.

EspHuifeng, 2020-08-18:
Have you set `(Top) -> Serial flasher config -> idf.py monitor' baud rate` and `(Top) -> Component config -> Common ESP-related -> UART console baud rate` both. 
Maybe the chip is working on 921600 baudrate but the serial tools is working on 115200 baudrate.

lybaohuy, 2020-08-18:
i use a other terminate, not serial tool from ESP
I make sure that the terminate's baudrate is 921600

EspHuifeng, 2020-08-26:
Can you give us more information: which example do you use, mdf commit id and sdkconfig file

lybaohuy, 2020-08-26:
here is example that used .
esp-mdf/examples/function_demo/mwifi/router/main/router_example.c

I changed UART0 console's baudrate in menuconfig to 912600.

on release1.0, it work well
but on master branch, it just work at first boot time but its baudrate is 115200( not 912600), and after that on terminal cann't see correct characters on 912600, even on other baudrates.

EspHuifeng, 2020-09-09:
Can you put your sdkconfig here?