espressif/esp-mdf #166: ERROR: non-ROOT nodes also receive "MDF_EVENT_MWIFI_ROOT_GOT_IP" event

lybaohuy, 2020-08-18:
I detected that non-root nodes also receive "MDF_EVENT_MWIFI_ROOT_GOT_IP" event,
i think that it's is wrong, pls check it

esp-mdf/examples/function_demo/mwifi/router/main/router_example.c 

static mdf_err_t event_loop_cb(mdf_event_loop_t event, void *ctx)
{
    MDF_LOGI("event_loop_cb, event: %d", event);

    switch (event) {
        case MDF_EVENT_MWIFI_STARTED:
            MDF_LOGI("MESH is started");
            break;

        case MDF_EVENT_MWIFI_PARENT_CONNECTED:
            MDF_LOGI("Parent is connected on station interface");

            if (esp_mesh_is_root()) {
                esp_netif_dhcpc_start(netif_sta);
            }

            break;

        case MDF_EVENT_MWIFI_PARENT_DISCONNECTED:
            MDF_LOGI("Parent is disconnected on station interface");
            break;

        case MDF_EVENT_MWIFI_ROUTING_TABLE_ADD:
        case MDF_EVENT_MWIFI_ROUTING_TABLE_REMOVE:
            MDF_LOGI("total_num: %d", esp_mesh_get_total_node_num());
            break;

        case MDF_EVENT_MWIFI_ROOT_GOT_IP: {
            MDF_LOGI("Root obtains the IP address. It is posted by LwIP stack automatically");
            xTaskCreate(tcp_client_write_task, "tcp_client_write_task", 4 * 1024,
                        NULL, CONFIG_MDF_TASK_DEFAULT_PRIOTY, NULL);
            xTaskCreate(tcp_client_read_task, "tcp_server_read", 4 * 1024,
                        NULL, CONFIG_MDF_TASK_DEFAULT_PRIOTY, NULL);
            break;

EspHuifeng, 2020-08-26:
Can you provide log and mdf commit to us , or tell us how to reproduce it.

lybaohuy, 2020-08-26:
on master branch , It is ok now
last time when i tested it on release1.0, it always occur on non-root nodes. I don't remember mdf commit because i updated to master branch few days ago.

You can try it on release1.0 at esp-mdf/examples/function_demo/mwifi/router/main/router_example.c.
i tested it on 3 nodes. 2 non-root alway occur "MDF_EVENT_MWIFI_ROOT_GOT_IP" event.

EspHuifeng, 2020-09-09:
I have try 4 nodes on commit 66f85308，and three non-root nodes won't occur event: 17(MDF_EVENT_MWIFI_ROOT_GOT_IP)