espressif/esp-mdf #168: error: 'CONFIG_BOOTLOADER_OFFSET_IN_FLASH' undeclared (first use in this function);

ljb00ne, 2020-08-28:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-MDF Programming Guide](https://docs.espressif.com/projects/esp-mdf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my MDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-mdf/en/latest/get-started/mdf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-mdf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM32E]
- MDF version (b4dd533.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		[VSCode]
- Operating System:		[Ubuntu|MacOS]
- Power Supply:         [USB]

## Problem Description

In trying to compile the router sample code from the /esp/esp-mdf/examples/function_demo/mwifi/router repo I am getting the following errors shown in debug logs.

This is the latest versions of both frameworks as of 8/28/20.

### Expected Behavior

This is example code and a fresh installation of the esp-idf, and esp-mdf. I have not been able to find anyone else with these issues.

### Actual Behavior

Shown above, does not compile.

### Steps to repropduce

 ljboone@LJs-Macbook-Pro router %  . ~/esp/esp-idf/export.sh
Setting IDF_PATH to '/Users/ljboone/esp/esp-idf'
Adding ESP-IDF tools to PATH...
Using Python interpreter in /Users/ljboone/.espressif/python_env/idf4.3_py2.7_env/bin/python
Checking if Python packages are up to date...
Python requirements from /Users/ljboone/esp/esp-idf/requirements.txt are satisfied.
Updated PATH variable:
  /Users/ljboone/esp/esp-idf/components/esptool_py/esptool:/Users/ljboone/esp/esp-idf/components/espcoredump:/Users/ljboone/esp/esp-idf/components/partition_table:/Users/ljboone/esp/esp-idf/components/app_update:/Users/ljboone/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin:/Users/ljboone/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin:/Users/ljboone/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/ljboone/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/ljboone/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin:/Users/ljboone/.espressif/python_env/idf4.3_py2.7_env/bin:/Users/ljboone/esp/esp-idf/tools:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/Library/Apple/bin:/Applications/VMware Fusion.app/Contents/Public:/usr/local/share/dotnet:/opt/X11/bin:~/.dotnet/tools:/Library/Frameworks/Mono.framework/Versions/Current/Commands
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

ljboone@LJs-Macbook-Pro router % idf.py build

### Code to reproduce this issue

example router code, no changes except for menuconfig network settings.

[... 26 lines cut ...]
 #define ESP_BOOTLOADER_OFFSET CONFIG_BOOTLOADER_OFFSET_IN_FLASH  /* Offset of bootloader image. Has matching value in bootloader KConfig.projbuild file. */
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[7/17] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_flash_config_esp32.c.obj
[8/17] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/bootloader_console.c.obj
[9/17] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32/flash_encrypt.c.obj
FAILED: esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32/flash_encrypt.c.obj 
/Users/ljboone/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -Iconfig -I/Users/ljboone/esp/esp-idf/components/bootloader_support/include -I/Users/ljboone/esp/esp-idf/components/bootloader_support/include_bootloader -I/Users/ljboone/esp/esp-idf/components/log/include -I/Users/ljboone/esp/esp-idf/components/esp_rom/include -I/Users/ljboone/esp/esp-idf/components/esp_common/include -I/Users/ljboone/esp/esp-idf/components/xtensa/include -I/Users/ljboone/esp/esp-idf/components/xtensa/esp32/include -I/Users/ljboone/esp/esp-idf/components/esp32/include -I/Users/ljboone/esp/esp-idf/components/soc/src/esp32/. -I/Users/ljboone/esp/esp-idf/components/soc/src/esp32/include -I/Users/ljboone/esp/esp-idf/components/soc/include -I/Users/ljboone/esp/esp-idf/components/soc/soc/esp32/. -I/Users/ljboone/esp/esp-idf/components/soc/soc/esp32/include -I/Users/ljboone/esp/esp-idf/components/soc/soc/esp32/../include -I/Users/ljboone/esp/esp-mdf/esp-idf/components/bootloader/subproject/components/micro-ecc -I/Users/ljboone/esp/esp-mdf/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc -I/Users/ljboone/esp/esp-idf/components/spi_flash/include -I/Users/ljboone/esp/esp-idf/components/efuse/include -I/Users/ljboone/esp/esp-idf/components/efuse/esp32/include -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -fno-stack-protector -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3-dev-907-g6c17e3a64\" -DESP_PLATFORM -DBOOTLOADER_BUILD=1 -MD -MT esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32/flash_encrypt.c.obj -MF esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32/flash_encrypt.c.obj.d -o esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp32/flash_encrypt.c.obj -c /Users/ljboone/esp/esp-idf/components/bootloader_support/src/esp32/flash_encrypt.c
In file included from /Users/ljboone/esp/esp-idf/components/bootloader_support/include/esp_image_format.h:18,
                 from /Users/ljboone/esp/esp-idf/components/bootloader_support/src/esp32/flash_encrypt.c:18:
/Users/ljboone/esp/esp-idf/components/bootloader_support/src/esp32/flash_encrypt.c: In function 'encrypt_bootloader':
/Users/ljboone/esp/esp-idf/components/bootloader_support/include/esp_flash_partitions.h:47:31: error: 'CONFIG_BOOTLOADER_OFFSET_IN_FLASH' undeclared (first use in this function); did you mean 'CONFIG_BOOTLOADER_WDT_ENABLE'?
 #define ESP_BOOTLOADER_OFFSET CONFIG_BOOTLOADER_OFFSET_IN_FLASH  /* Offset of bootloader image. Has matching value in bootloader KConfig.projbuild file. */
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/ljboone/esp/esp-idf/components/bootloader_support/include/esp_flash_partitions.h:47:31: note: in definition of macro 'ESP_BOOTLOADER_OFFSET'
 #define ESP_BOOTLOADER_OFFSET CONFIG_BOOTLOADER_OFFSET_IN_FLASH  /* Offset of bootloader image. Has matching value in bootloader KConfig.projbuild file. */
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/ljboone/esp/esp-idf/components/bootloader_support/include/esp_flash_partitions.h:47:31: note: each undeclared identifier is reported only once for each function it appears in
[... cut ...] .projbuild file. */
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/ljboone/esp/esp-idf/components/bootloader_support/include/esp_flash_partitions.h:47:31: note: in definition of macro 'ESP_BOOTLOADER_OFFSET'
 #define ESP_BOOTLOADER_OFFSET CONFIG_BOOTLOADER_OFFSET_IN_FLASH  /* Offset of bootloader image. Has matching value in bootloader KConfig.projbuild file. */

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-mdf/files/5142671/sdkconfig.txt)

closed 2020-09-04 by ljb00ne

baybal, 2020-09-17:
@ljb00ne how did you solve the issue?

ljb00ne, 2020-09-20:
@baybal I created a dedicated VM for the idf and reinstalled the dependencies. I don’t have a specific solution.