espressif/esp-mdf #170: idf.py build fail

miteshdube, 2020-09-21:
I am new to ESP-MDF. I'm trying to run basic get-started example in ESP-MDF but its giving error as below, (Note: idf.py menuconfig running successfully.)

```
mitesh@mitesh-Inspiron-3542:~/esp/esp-mdf/get-started$ idf.py build
Setting IDF_PATH environment variable: /home/mitesh/esp/esp-mdf/esp-idf
Executing action: all (aliases: build)
Running ninja in directory /home/mitesh/esp/esp-mdf/get-started/build
Executing "ninja all"...
ninja: error: '../secure_boot_signing_key.pem', needed by 'signature_verification_key.bin', missing and no known rule to make it
ninja failed with exit code 1
```

EspHuifeng, 2020-09-23:
You may turn on the options in security-features, you can run idf.py  menuconfig to check it.

Lss615, 2021-03-12:
I have similar questions.First,when I use the git to download the MD F,it shows as follows:
[image]
There are some files that it can't get.I have tried many times but failed.
In addition,when I'm trying to run basic get-started example,it shows
[image]
[image]

closed 2021-08-23 by miteshdube