espressif/esp-mdf #172: Error in uploding configuration through ESP-Mesh-App in Light example development_kit.

miteshdube, 2020-09-28:
I followed all procedure given on this [link](https://github.com/espressif/esp-mdf/tree/master/examples/development_kit). But whenever I tried to add device it will upload configuration till 25% and after that I am getting below error.

[image]
[image]

Hiren-Sorathiya, 2020-09-30:
Please check the Mesh password length.

EspHuifeng, 2020-10-28:
Hi @miteshdube ，Are you using the latest IDF version

miteshdube, 2020-11-01:
Here is my IDF version
```
mdf> version
I (10073) [mdebug_cmd, 53]: ESP-IDF version  : v4.3-dev-1197-g8bc19ba-dirty
I (10074) [mdebug_cmd, 54]: ESP-MDF version  : v1.0-91-g00f7df4-dirty
I (10085) [mdebug_cmd, 55]: compile time     : Nov  1 2020 11:24:10
I (10086) [mdebug_cmd, 56]: free heap        : 121376 Bytes
I (10097) [mdebug_cmd, 57]: CPU cores        : 2
I (10098) [mdebug_cmd, 58]: silicon revision : 1
I (10109) [mdebug_cmd, 64]: feature          : /802.11bgn/BLE/BT/External-Flash:4 MB
mdf> 

```

EspHuifeng, 2020-11-02:
This is a small problem with bt in the recent IDF, it would be fixed later. Can you use the version specified by submodule in MDF.

miteshdube, 2020-11-02:
Okay I'll check with this solution and Thank you.

miteshdube, 2020-11-17:
I cloned esp-idf version as per mentioned in this [link](https://github.com/espressif/esp-mdf) still getting same error. Below is the new changed version  @EspHuifeng 
```
mdf> version
I (27481) [mdebug_cmd, 53]: ESP-IDF version  : v4.3-dev-1720-g494a124
I (27482) [mdebug_cmd, 54]: ESP-MDF version  : v1.0-91-g00f7df4-dirty
I (27493) [mdebug_cmd, 55]: compile time     : Nov 17 2020 23:22:11
I (27494) [mdebug_cmd, 56]: free heap        : 101704 Bytes
I (27505) [mdebug_cmd, 57]: CPU cores        : 2
I (27505) [mdebug_cmd, 58]: silicon revision : 1
I (27516) [mdebug_cmd, 64]: feature          : /802.11bgn/BLE/BT/External-Flash:4 MB
mdf> 
```

jayjayhust, 2021-02-24:
hey bro, try this release version: https://github.com/espressif/esp-mdf/releases/tag/v1.0

dont use other MDF or IDF version.

closed 2021-08-23 by miteshdube