espressif/esp-mdf #177: AWS IoT example

andresv, 2020-10-22:
I can see from README that AWS IoT example is in to be released state.
What is status with AWS, is it technically already usable and just example is missing or is there some core components missing?

esp-lqq, 2021-05-25:
Hi andresv
The project is currently in the development plan，Due to the current shortage of resources, it has not yet been scheduled.

andresv, 2021-05-25:
Could you briefly tell me what is actually not yet implemented to get it working?

andresv, 2021-06-03:
More information about this topic: https://esp32.com/viewtopic.php?f=21&t=21084

esp-lqq, 2021-06-03:
Hi andresv
 we plan to start supporting the construction of the MDF ip layer protocol in the third quarter, and subsequent MDF framework devices will use IDF upper-level components, such as AWS, like the normal routing mode.

andresv, 2022-01-09:
Do you have any updates about timeline for this feature?