espressif/esp-mdf #178: how to send/receive group package through the wifi-mesh without the router

NOA-Vernon, 2020-10-23:
mwifi_init_config_t cfg   = MWIFI_INIT_CONFIG_DEFAULT();
#if 0
	mwifi_config_t config     = {
		.router_ssid     = CONFIG_ROUTER_SSID,
		.router_password = CONFIG_ROUTER_PASSWORD,
		.channel  		 = CONFIG_MESH_CHANNEL,
		.mesh_id         = CONFIG_MESH_ID,
		.mesh_password   = CONFIG_MESH_PASSWORD,
		//.mesh_type = MWIFI_MESH_ROOT,
		.mesh_type = MWIFI_MESH_NODE,
	};
#else
	mwifi_config_t config;
	memset(&config, 0x00, sizeof(mwifi_config_t));

	config.channel  		 = CONFIG_MESH_CHANNEL;
	//config.mesh_id         = CONFIG_MESH_ID;
	//config.mesh_password   = CONFIG_MESH_PASSWORD;
	memcpy( config.mesh_id, &nvs_store.group_uuid[0], 6 );
	memcpy( config.mesh_password, &nvs_store.group_uuid[0], strlen(nvs_store.group_uuid) );

	if( nvs_store.dev_character[0] == '1' ){
		memcpy( config.router_ssid, "abcdefg", strlen("abcdefg") );
		memcpy( config.router_password, "axaxaxaxax", strlen("axaxaxaxax") );
		//memcpy( config.router_ssid, "NOA Labs (2.4GHz)", strlen("NOA Labs (2.4GHz)") );
		//memcpy( config.router_password, "noa-labs.com", strlen("noa-labs.com") );

		config.mesh_type = MWIFI_MESH_ROOT;
	}else{
		config.mesh_type = MWIFI_MESH_NODE;
		//config.mesh_type = MWIFI_MESH_IDLE;
	}

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-MDF Programming Guide](https://docs.espressif.com/projects/esp-mdf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my MDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-mdf/en/latest/get-started/mdf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-mdf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Core (if using chip or module): [ESP32-Wrover-I|ESP32-Wrover|ESP32-PICO-D4|ESP32-SOLO-I|ESP32-WROOM-32U|ESP-WROOM-32D|ESP-WROOM32|ESP32]
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		[Arduino IDE|Make|Eclipse|other]
- Operating System:		[Windows|Ubuntu|MacOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

EspHuifeng, 2020-10-23:
What means "send/receive group package without router", Can you describe your question and scenario in more detail.

NOA-Vernon, 2020-10-23:
@EspHuifeng I have updated the question, I mis-click the button just now, can you please answer my question above? Thank you very much.

NOA-Vernon, 2020-10-26:
@EspHuifeng Is there any updates here?

EspHuifeng, 2020-10-26:
Hi @NOA-Vernon ，I have tested mesh group function and it can work well. Do you invoke mwifi_read() function in your program?

NOA-Vernon, 2020-10-26:
yes, I did really call the mwifi_read() function in the node_read_task task, are you testing without a  router?

EspHuifeng, 2020-10-26:
You are right, I reproduce this issue in fix-root mode without a router. we will find why and fix it as soon as possible.

NOA-Vernon, 2020-10-26:
@EspHuifeng  And if the esp32 try to work at wifi-mesh mode without a router, the router_ssid and router_password shouldn't be configured, see the following snippet code

	mwifi_config_t config     = {
//		.router_ssid     = CONFIG_ROUTER_SSID,
//		.router_password = CONFIG_ROUTER_PASSWORD,
		.channel  		 = CONFIG_MESH_CHANNEL,
		.mesh_id         = CONFIG_MESH_ID,
		.mesh_password   = CONFIG_MESH_PASSWORD,
		//.mesh_type = MWIFI_MESH_ROOT,
		.mesh_type = MWIFI_MESH_NODE,
	};
If set the router_ssid and router_password, the data transmission will delay for a few seconds when try to send a package to another device by calling the function mwifi_write(mac, &data_type, data, len, true); it seems that the esp32 is always scanning the router and try to associate, can you test this issue as well?

EspHuifeng, 2020-10-28:
@NOA-Vernon, If there is no router in fix-root situation, the mwifi_read() function can not be run by the root node in node_read_task because mwifi_is_connected() won't return true. This is why that group message can not be received by another node which is the same group id. 
As for another question. You should not set an incorrect SSID and password. The root node will change the channel for scan AP, This action can be last until the root gets connected.

NOA-Vernon, 2020-10-28:
@EspHuifeng  Thanks for your answer, so are you going to fix the group-cast without a router recently? I have some devices that work at the wifi-mesh network and associate with the router correctly, however, when I bring these devices to go outside and stay far away the router, the device can't communicate in time or there are some delays during the communication, how can I fix this issue, do you have any suggestions?

EspHuifeng, 2020-10-28:
This is not a bug and no need to fix. If you use mesh in no-router situation. You can reference this code: https://github.com/espressif/esp-mdf/blob/e5240f182433ae0c73a8d3ced64fdbb3f1e8ce41/examples/function_demo/mwifi/no_router/main/no_router_example.c#L144

NOA-Vernon, 2020-11-20:
@EspHuifeng I have another question for you, in the no router wifi mesh, I want to enable BLE as well for communicating with the app. However, When I commented 
//MDF_ERROR_ASSERT(esp_wifi_set_ps(WIFI_PS_NONE)); 
the BLE and wifi mesh can coexist and run, but the root and node device can't communicate, neither the node end to root device nor root device send to node device, the receiving device can't receive from the sending device, you can test it for confirmation.

Jigar3690, 2021-01-26:
@EspHuifeng I tried exact same set up what @NOA-Vernon had.
I initially used esp-mesh-send() function but had no luck may be because I am using it with mdf that's why.

Then I also tried with mwifi_write function - still no luck.
I am able to send the data to the group but nodes are not able to receive anything.
In my case I am trying to do this with router connected - and still it's not going through.

I am having everything same as @NOA-Vernon description. Any idea what could be wrong?

Jigar3690, 2021-01-26:
In fact, surprisingly - my groupID is different and when I try to write to another group ID (0x01, 0x00, 0x5e, 0xae, 0xae, 0xae) for which I haven't created any group yet!!! I am still getting no error while I use it with mwifi_write(). It still gives my MDF_OK in return.

So now I have doubt whether is my root sending data to group successfully or not. 
when I read the number of group then it increments as I create new groups, so I think I am able to create the groups successfully. Only not able to send anything to group. (also not receiving from group members)

sagar448, 2021-01-26:
@EspHuifeng Im having a similar problem here. Please also refer to this issue https://github.com/espressif/esp-mdf/issues/194

NaimSakaamini, 2021-01-27:
any fix for this yet? I'm facing the same issue!