espressif/esp-mdf #179: Is it feasible to use raspberry pi to connect to wifi mesh without router？

huang443765159, 2020-10-23:
Hello.
 I'm running a test for the company.  We want to use esp32 to detect sensor data and use wifi mesh to transmit the data to raspberry pi.
I saw that using the router to connect to the root node could communicate with the external server ip, but we don't want to use the router, because the router will have dropped calls and misconnections.
Is there a way to connect a raspberry pie to Mesh wifi without using router？
Thanks.

EspHuifeng, 2020-10-28:
You can let RPi as a softAP and the root can connect to RPi.

huang443765159, 2020-10-28:
> 您可以让RPi作为softAP，并且根可以连接到RPi。

Thanks for your help!
 I will try to do it.

JL1946, 2021-09-07:
hello:
I experimented with using the raspberry pi in wifi-softAP and wifi-client mode but was not happy with the result due to channel selection... I then tried with a separate wifi dongle plugged into the USB on the pi, with limited success due to problems caused by the linux wifi dongle drivers... I am now using an esp32-wroom dev kit as a "Gateway" which is setup as a softAP and acts as the "router" for the mesh and communicates with the pi over SPI...