espressif/esp-mdf #181: Getting timeout issue in mupgrade_example

techyx, 2020-10-28:
As stated in README.md file, followed same steps > [](https://github.com/espressif/esp-mdf/blob/master/examples/function_demo/mupgrade/README.md)

Only change that i made is instead of writing below line 
`python -m SimpleHTTPServer 8070`
I tried 
`python -m http.server 8070`

esp mdf git branch > master

Below serial log given 
`I (21322) [mupgrade_example, 132]: Open HTTP connection: http://0.0.0.0:8070/Blink.ino.esp32.bin
E (26327) TRANS_TCP: [sock=54] select() timeout
E (26328) HTTP_CLIENT: Connection failed, sock < 0
W (27328) [mupgrade_example, 146]: <ESP_ERR_HTTP_CONNECT> Connection service failed
E (32334) TRANS_TCP: [sock=54] select() timeout
E (32335) HTTP_CLIENT: Connection failed, sock < 0
`

below cmd log
`curl -v http://0.0.0.0:8070/Blink.ino.esp32.bin`

`*   Trying 0.0.0.0:8070...
* TCP_NODELAY set
* Connected to 0.0.0.0 (127.0.0.1) port 8070 (#0)
> GET /Blink.ino.esp32.bin HTTP/1.1
> Host: 0.0.0.0:8070
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: SimpleHTTP/0.6 Python/3.8.5
< Date: Wed, 28 Oct 2020 19:14:07 GMT
< Content-type: application/octet-stream
< Content-Length: 207824
< Last-Modified: Wed, 28 Oct 2020 18:27:22 GMT
< 
Warning: Binary output can mess up your terminal. Use "--output -" to tell 
Warning: curl to output it to your terminal anyway, or consider "--output 
Warning: <FILE>" to save to a file.
* Failed writing body (0 != 16384)
* Closing connection 0
`

EspHuifeng, 2020-10-29:
Hi @techyx. You must set a correct HTTP URL in menuconfig. "http://0.0.0.0:8070/Blink.ino.esp32.bin" is not correct. 
```txt
I (21322) [mupgrade_example, 132]: Open HTTP connection: http://0.0.0.0:8070/Blink.ino.esp32.bin
```

techyx, 2020-10-29:
Sorry, newbie here
Thanks for hint
Solved

closed 2020-10-29 by techyx