espressif/esp-mdf #182: ESP Mesh APP Android and Mconfig example

slyfoxneil, 2020-11-13:
Hello,

I trying the mconfig example from https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mconfig and installed the android app esp mesh.

the device log is showing this

I (870) cpu_start: Starting scheduler on PRO CPU.
I (903) wifi:wifi driver task: 3ffcce94, prio:23, stack:6656, core=0
I (904) system_api: Base MAC address is not set
I (904) system_api: read default base MAC address from EFUSE
I (917) wifi:wifi firmware version: 8ee1f07
I (917) wifi:wifi certification version: v7.0
I (918) wifi:config NVS flash: enabled
I (921) wifi:config nano formating: disabled
I (925) wifi:Init data frame dynamic rx buffer num: 32
I (930) wifi:Init management frame dynamic rx buffer num: 32
I (935) wifi:Init management short buffer num: 32
I (940) wifi:Init dynamic tx buffer num: 32
I (944) wifi:Init static rx buffer size: 1600
I (948) wifi:Init static rx buffer num: 10
I (952) wifi:Init dynamic rx buffer num: 32
I (1038) phy: phy_version: 4370, 4e803b3, Aug 11 2020, 14:18:07, 0, 0
I (1040) wifi:mode : sta (98:f4:ab:72:2b:0c)
I (1042) ESPNOW: espnow [version: 1.0] init
I (1043) [mconfig_example, 112]: BLE name: ESP-WIFI-MESH_2b0c
I (1048) [mconfig_example, 52]: event_loop_cb, event: 8704
I (1975) BTDM_INIT: BT controller compile version [e250020]
I (2355) [mconfig_blufi, 541]: BLUFI init finish, set ble advertising data
I (2359) [mconfig_example, 52]: event_loop_cb, event: 8449
I (2362) [mconfig_blufi, 884]: start ble advertising
I (2880) [mconfig_chain, 361]: Generate RSA public and private keys
I (2881) wifi:ic_enable_sniffer

But the esp mesh app can not detect the device. The app is always displaying I am not connected the the wifi access point/router even though I am connected.
[image]

Any help is very much appreciated. 

Thanks in advance

Neil Ablang

Jiangyafeng, 2020-11-13:
Please turn on the location of your mobile phone, then restart and open the app.

Jiangyafeng, 2020-11-13:
@slyfoxneil  Please turn on the location of your mobile phone, then restart and open the app.

slyfoxneil, 2020-11-13:
Hello JIanyafeng,

Thanks, it works!

My problem is reading the succeeding pages. They are in Chinese and I can't
read them
[image: tempFileForShare_20201113-193954.jpg]

Where can I download the Android mobile app source codes so that I'll try
to modify it?

Thanks again

Neil Ablang

Jiangyafeng, 2020-11-17:
@slyfoxneil 

Please change the language of the mobile phone to English mode, and you can see that the mesh app will also change, so you can understand it. 

For mobile mesh app application, the source code link is as follows: https://github.com/EspressifApp/EspMeshForiOS

Jiangyafeng, 2020-11-17:
@slyfoxneil For the Chinese language in the configuration, it's just an embedded image. You don't need to worry about it. When configuring, you can directly enter the WiFi and password you want to link to.

adokitkat, 2021-02-25:
Hello, I have similar problem. I am trying to run mconfig example, but in Esp Mesh App v1.2.3  on Android 10 it just cannot find any Bluetooth device - wifi, bluetooth and location is all on. The app just doesn't see ESP32-Buddy.

Monitor output:
```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7144
load:0x40078000,len:13704
load:0x40080400,len:4000
0x40080400: _init at ??:?

entry 0x40080688
I (28) boot: ESP-IDF v4.2-dev-2084-g98d5b5dfd-dirty 2nd stage bootloader
I (28) boot: compile time 01:12:18
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (100) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (107) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (115) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (123) boot: End of partition table
I (127) boot: No factory image, trying OTA 0
I (132) boot_comm: chip revision: 1, min. application chip revision: 0
I (139) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3b3fc (242684) map
I (240) esp_image: segment 1: paddr=0x0004b424 vaddr=0x3ffbdb60 size=0x04bf4 ( 19444) load
I (249) esp_image: segment 2: paddr=0x00050020 vaddr=0x400d0020 size=0x104148 (1065288) map
0x400d0020: _stext at ??:?

I (654) esp_image: segment 3: paddr=0x00154170 vaddr=0x3ffc2754 size=0x001bc (   444) load
I (655) esp_image: segment 4: paddr=0x00154334 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/ado/esp/esp-mdf/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (662) esp_image: segment 5: paddr=0x00154740 vaddr=0x40080404 size=0x1ca3c (117308) load
I (736) boot: Loaded app from partition at offset 0x10000
I (755) boot: Set actual ota_seq=1 in otadata[0]
I (755) boot: Disabling RNG early entropy source...
I (755) cpu_start: Pro cpu up.
I (758) cpu_start: Application information:
I (763) cpu_start: Project name:     mconfig
I (768) cpu_start: App version:      c773ee4-dirty
I (774) cpu_start: Compile time:     Feb 25 2021 01:12:10
I (780) cpu_start: ELF file SHA256:  64f2f39381dbd182...
I (786) cpu_start: ESP-IDF:          v4.2-dev-2084-g98d5b5dfd-dirty
I (793) cpu_start: Single core mode
I (797) heap_init: Initializing. RAM available for dynamic allocation:
I (804) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (810) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (816) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (822) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (829) heap_init: At 3FFCA748 len 000158B8 (86 KiB): DRAM
I (835) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (841) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (847) heap_init: At 4009CE40 len 000031C0 (12 KiB): IRAM
I (853) cpu_start: Pro cpu start user code
I (871) spi_flash: detected chip: gd
I (872) spi_flash: flash io: dio
W (872) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (882) esp_core_dump_flash: Init core dump to flash
I (888) esp_core_dump_flash: Found partition 'coredump' @ 3d0000 65536 bytes
I (896) cpu_start: Starting scheduler on PRO CPU.
I (929) wifi:wifi driver task: 3ffcce88, prio:23, stack:6656, core=0
I (930) system_api: Base MAC address is not set
I (930) system_api: read default base MAC address from EFUSE
I (942) wifi:wifi firmware version: 8ee1f07
I (943) wifi:wifi certification version: v7.0
I (944) wifi:config NVS flash: enabled
I (947) wifi:config nano formating: disabled
I (951) wifi:Init data frame dynamic rx buffer num: 32
I (956) wifi:Init management frame dynamic rx buffer num: 32
I (961) wifi:Init management short buffer num: 32
I (966) wifi:Init dynamic tx buffer num: 32
I (970) wifi:Init static rx buffer size: 1600
I (974) wifi:Init static rx buffer num: 10
I (978) wifi:Init dynamic rx buffer num: 32
I (1067) phy: phy_version: 4370, 4e803b3, Aug 11 2020, 14:18:07, 0, 0
I (1069) wifi:mode : sta (3c:71:bf:e4:8f:00)
I (1071) ESPNOW: espnow [version: 1.0] init
I (1072) [mconfig_example, 110]: BLE name: ESP-WIFI-MESH_8f00
I (1077) [mconfig_example, 50]: event_loop_cb, event: 8704
I (2430) BTDM_INIT: BT controller compile version [e250020]
I (2783) [mconfig_blufi, 676]: BLUFI init finish, set ble advertising data
I (2787) [mconfig_example, 50]: event_loop_cb, event: 8449
I (2790) [mconfig_blufi, 1069]: start ble advertising
I (5426) [mconfig_chain, 369]: Generate RSA public and private keys
I (5428) wifi:ic_enable_sniffer
```

Lss615, 2021-03-12:
1. I used the Mesh APP to connect with my  ESP-Buddy and lamp.But in the process of distributing network,when it reached 60%，it'd remain unchanged.Then,it shows Bluetooth connection is broken while my bluetooth  go smoothly.
[image]
I'd been confused a long time.

manikandan341, 2023-11-09:
Hello, I'm currently utilizing the "ble_mesh_wifi_coexist" example. I've successfully established a serial communication connection with my primary router. However, I'm encountering challenges in configuring both the server and client. Additionally, there are no devices appearing in the MeshKit application, although my device is visible in the "NRF Mesh" Application. Can you offer guidance on addressing these configuration issues and ensuring proper device visibility in the MeshKit application?

tonyshuang, 2023-11-09:
您好，信件已收到。

manikandan341, 2023-11-09:
Please help me now...