espressif/esp-mdf #183: 移植阿里云 生活物联网平台SDK（V1.6.2）到Mesh_witch_aliyun

kust336, 2020-11-18:
你好!
  请问下有计划适配 生活物联网平台SDK（V1.6.2）的功能吗? 实现云智能公版APP配网及控制.

Jiangyafeng, 2020-11-26:
@kust336 mdf 里面使用自己的一套程序实现了云智能公版 APP 配网及控制，但是没有使用生活物联网平台的SDK，具体链接可以参考：https://github.com/espressif/esp-mdf/tree/master/components/third_party/aliyun_sdk，例程可以参看mdf 下的example。

kust336, 2020-11-26:
aliyun_sdk 是配合 examples/aliyun_linkit例程使用的吗?
 看了下get-started   mesh-with-aliyun   都是使用固定WIFI  ssid和pawssd 没看到有关于云智能公版 APP 配网有例程.
请帮助我了解下mdf下使用云智能公版 APP 配网及控制.  谢谢

esp-lqq, 2020-11-27:
1. 当前的 MDF 中的 aliyun sdk 没有移植公版的云智能 APP 的配网，默认使用 MESH APP 的蓝牙配网。
2. 可以使用将 esp-aliyun 中的 wifi 配网模块移植使用，实现公版 APP 配网的能力。

jayjayhust, 2021-02-17:
> 1. 当前的 MDF 中的 aliyun sdk 没有移植公版的云智能 APP 的配网，默认使用 MESH APP 的蓝牙配网。
> 2. 可以使用将 esp-aliyun 中的 wifi 配网模块移植使用，实现公版 APP 配网的能力。

嗨，请问MDF 中的 aliyun sdk ，使用MESH APP的蓝牙配网的示例或者代码部分主要是在哪个部分呢？我找了一下aliyun sdk，好像没看到啊，有劳大神指点迷津啊~~~

kust336, 2021-02-18:
> > 1. 当前的 MDF 中的 aliyun sdk 没有移植公版的云智能 APP 的配网，默认使用 MESH APP 的蓝牙配网。
> > 2. 可以使用将 esp-aliyun 中的 wifi 配网模块移植使用，实现公版 APP 配网的能力。
> 
> 嗨，请问MDF 中的 aliyun sdk ，使用MESH APP的蓝牙配网的示例或者代码部分主要是在哪个部分呢？我找了一下aliyun sdk，好像没看到啊，有劳大神指点迷津啊~~~

closed 2021-02-18 by kust336

kust336, 2021-02-18:
> > 1. 当前的 MDF 中的 aliyun sdk 没有移植公版的云智能 APP 的配网，默认使用 MESH APP 的蓝牙配网。
> > 2. 可以使用将 esp-aliyun 中的 wifi 配网模块移植使用，实现公版 APP 配网的能力。
> 
> 嗨，请问MDF 中的 aliyun sdk ，使用MESH APP的蓝牙配网的示例或者代码部分主要是在哪个部分呢？我找了一下aliyun sdk，好像没看到啊，有劳大神指点迷津啊~~~

在 esp-mdf/examples/development_kit/light. 
入口函数: get_network_config(&init_config, &ap_config, LIGHT_TID, LIGHT_NAME)

jayjayhust, 2021-02-19:
> 在 esp-mdf/examples/development_kit/light.
> 入口函数: get_network_config(&init_config, &ap_config, LIGHT_TID, LIGHT_NAME)

已经找到，应用跑起来了，非常感谢兄弟指点！