espressif/esp-mdf #184: mwifi_write的阻塞问题

fzboffice, 2020-11-19:
在高频调用mwifi_write的时候 不规律地出现阻塞 最终锁定在mwifi_write->mwifi_subcontract_write->esp_mesh_send 这个函数
阻塞几乎都是3秒 返回的是ESP_OK 但是我的项目需求不允许阻塞 我想知道该怎么样才可以不阻塞 因为mwifi_write里有一个参数可以选择不阻塞 但是实际情况还是阻塞了 我的mdf是'release/v1.0'

Jiangyafeng, 2020-11-25:
@fzboffice mesh上行发送数据是存在流控的，所以会存在阻塞状态。具体链接可以参考：https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/mesh.html。

fzboffice, 2020-11-26:
> @fzboffice mesh上行发送数据是存在流控的，所以会存在阻塞状态。具体链接可以参考：[https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/mesh.html。](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/mesh.html%E3%80%82)

请问 我能做什么来取消这个流控或者减少阻塞时间吗

Jiangyafeng, 2020-11-27:
mesh 上行发送数据是存在流控，下行发送数据是可以关掉流控，也就是将其 block = false。如果想取消上行发送的流控，那就要采用广播形式，就可以取消发送的流控，但是这个存在问题是消息不能保证全部接收到。
广播设置：
uint8_t addr_any[]     = MWIFI_ADDR_BROADCAST;

 mwifi_data_type_t data_type = {
        .communicate = MWIFI_COMMUNICATE_BROADCAST,
        .group = false,
    };

mwifi_write(addr_any, &data_type, data, size, false);

这样发送可以取消流控限制。

fzboffice, 2020-11-27:
@Jiangyafeng 按照您这样说的去做 确实可以做到不阻塞了 我对于数据的到达保证性要求不高 但是广播终究还不是最佳解决方案，请问group发送 可以取消流控吗

Jiangyafeng, 2020-12-02:
@fzboffice 对于设置为group，也不能取消流控。你可以增大 xon_qsize 值，看是否可以减少阻塞时间。即在 menuconfig->componet config-> MDF Mwifi -> Transmission config 中的 the number of mesh buffer queues 的值。

fzboffice, 2021-02-24:
> @fzboffice 对于设置为group，也不能取消流控。你可以增大 xon_qsize 值，看是否可以减少阻塞时间。即在 menuconfig->componet config-> MDF Mwifi -> Transmission config 中的 the number of mesh buffer queues 的值。

但是奇怪的是，每一次发生阻塞都一定是3秒，我觉得挺不可思议的，到底为什么那么精确呢

fzboffice, 2021-06-04:
@Jiangyafeng 请问这个问题能帮助我解答一下吗 谢谢