espressif/esp-mdf #186: MDF_EVENT_MWIFI_ROOT_GOT_IP event does not occur on root device

PavelLot, 2020-12-02:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM-32U]
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 1c76187f
- Development Env:		[Eclipse]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

I found that sometimes when the root device goes offline, someone from the other child nodes on the network becomes the root node, but the MDF_EVENT_MWIFI_ROOT_GOT_IP event does not occur.
In this case, the esp_mesh_is_root method returns true.
Could there be conditions under which such behavior is possible.

aresstokrat, 2020-12-03:
I also have such a problem. Sometimes after voting is done ROOT does't get IP.
I tried different routers, but problem still occur. 

How can i debug this problem to find a solution?

closed 2020-12-17 by PavelLot

ravi-kashi, 2021-12-11:
I face the same issue. Has this been worked up on? I don't know why this is not working and the ticket is closed?

I found this in sdkconfig
`CONFIG_MWIFI_WAIVE_ROOT=y`

I found this in mwifi.c
```
#ifdef CONFIG_MWIFI_WAIVE_ROOT

        case IP_EVENT_STA_GOT_IP: {
            mwifi_waive_root_timer_delete();
            mwifi_waive_root_timer_create();
            mdf_event_loop_send(MDF_EVENT_MWIFI_ROOT_GOT_IP, &s_event_info);
            break;
        }

#endif /**< CONFIG_MWIFI_WAIVE_ROOT */
```

shenjun7, 2021-12-13:
Please update the following api, we will merge to the release version.
esp_err_t esp_netif_create_default_wifi_mesh_netifs(esp_netif_t **p_netif_sta, esp_netif_t **p_netif_ap)
{
    // Create "almost" default AP, with un-flagged DHCP server
    esp_netif_inherent_config_t netif_cfg;
    memcpy(&netif_cfg, ESP_NETIF_BASE_DEFAULT_WIFI_AP, sizeof(netif_cfg));
    netif_cfg.flags &= ~ESP_NETIF_DHCP_SERVER;
    esp_netif_config_t cfg_ap = {
            .base = &netif_cfg,
            .stack = ESP_NETIF_NETSTACK_DEFAULT_WIFI_AP,
    };
    esp_netif_t *netif_ap = esp_netif_new(&cfg_ap);
    assert(netif_ap);
    ESP_ERROR_CHECK(esp_netif_attach_wifi_ap(netif_ap));
    ESP_ERROR_CHECK(esp_wifi_set_default_wifi_ap_handlers());

    // ...and stop DHCP server to be compatible with former tcpip_adapter (to keep the ESP_NETIF_DHCP_STOPPED state)
    ESP_ERROR_CHECK(esp_netif_dhcps_stop(netif_ap));

    // Create "almost" default station, but with un-flagged DHCP client
    memcpy(&netif_cfg, ESP_NETIF_BASE_DEFAULT_WIFI_STA, sizeof(netif_cfg));
    // netif_cfg.flags &= ~ESP_NETIF_DHCP_CLIENT;
    esp_netif_config_t cfg_sta = {
            .base = &netif_cfg,
            .stack = ESP_NETIF_NETSTACK_DEFAULT_WIFI_STA,
    };
    esp_netif_t *netif_sta = esp_netif_new(&cfg_sta);
    assert(netif_sta);
    ESP_ERROR_CHECK(esp_netif_attach_wifi_station(netif_sta));
    ESP_ERROR_CHECK(esp_wifi_set_default_wifi_sta_handlers());

    // ...and stop DHCP client (to be started separately if the station were promoted to root)
    ESP_ERROR_CHECK(esp_netif_dhcpc_stop(netif_sta));

    if (p_netif_sta) {
        *p_netif_sta = netif_sta;
    }
    if (p_netif_ap) {
        *p_netif_ap = netif_ap;
    }
    return ESP_OK;
}

ravi-kashi, 2021-12-13:
Thank you Shenjun7.

the only difference is  `// netif_cfg.flags &= ~ESP_NETIF_DHCP_CLIENT;`
I now get a different error and not the root task is still not created.
`E (9466) esp_netif_handlers: invalid static ip`

shenjun7, 2021-12-16:
This error does not affect the acquisition of ip. You can add esp_netif_dhcpc_stop(sta_netif) after the MDF_EVENT_MWIFI_PARENT_CONNECTED, and revert the above change.

        case MDF_EVENT_MWIFI_PARENT_CONNECTED:
            MDF_LOGI("Parent is connected on station interface");

            if (esp_mesh_is_root()) {
                esp_netif_dhcpc_stop(sta_netif);
                esp_netif_dhcpc_start(sta_netif);
            }

ravi-kashi, 2021-12-16:
Thank you Shenjun7. 
I made these changes and it is working. I also removed the 

`esp_netif_dhcpc_stop(sta_netif); 
`

and checked. It is working all right. Appreciate the help!

xinort, 2022-04-27:
> Thank you Shenjun7. I made these changes and it is working. I also removed the
> 
> `esp_netif_dhcpc_stop(sta_netif); `
> 
> and checked. It is working all right. Appreciate the help!

@ravi-kashi Hi, can you explain the changes you made? I'm experiencing this issue and the instructions from @shenjun7 aren't clear to me what changes need to be made in what file(s).

piro-pp, 2022-09-19:
> @ravi-kashi Hi, can you explain the changes you made? I'm experiencing this issue and the instructions from @shenjun7 aren't clear to me what changes need to be made in what file(s).

It's the file esp-idf/components/esp_wifi/src/wifi_default.c around line 385. The only change is to outcomment the line "netif_cfg.flags &= ~ESP_NETIF_DHCP_CLIENT;"