espressif/esp-mdf #188: fatal error: esp_vfs_fat.h: No such file or directory

tatsutaigu, 2020-12-10:
fatal error: esp_vfs_fat.h: No such file or directory

## Environment

- Core (if using chip or module): ESP32-WROVER-B
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 
   1c76187
- Operating System:		WSL (Windows Subsystem on Linux)
- Power Supply:         USB

## Problem Description
I tried [function_demo/mwifi/router](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/router).
I would like to use SD card.
Then, I got below error.

> fatal error: esp_vfs_fat.h: No such file or directory

### Expected Behavior

### Actual Behavior

### Steps to repropduce
set up ESP-MDF
`cd $MDF_PATH`
`./install.sh`
`. ./export.sh`

`idf.py menuconfig`
`idf.py build`

### Code to reproduce this issue
add below code to router_example.c
```
#include "esp_vfs_fat.h"
#include "driver/sdspi_host.h"
#include "driver/spi_common.h"
#include "sdmmc_cmd.h"
#include "driver/sdmmc_host.h"
```

tatsutaigu, 2020-12-15:
I modified CMakeList.txt.
Then, it was resolved.

closed 2020-12-15 by tatsutaigu

jonli1, 2021-02-01:
> I modified CMakeList.txt.
> Then, it was resolved.

How modified?

jonli1, 2021-02-01:
> > I modified CMakeList.txt.
> > Then, it was resolved.
> 
> How modified?
add components fatfs
it was resolved.

tatsutaigu, 2021-02-01:
Yes.
I added fatfs.

tatsutaigu, 2022-10-06:
1. open and modify `$YOUR_PROJECT/main/CMakeList.txt`.
2. add `fatfs vfs sdmmc` as below.
set(COMPONENT_REQUIRES mcommon mconfig mwifi mlink fatfs vfs sdmmc)