espressif/esp-mdf #189: error when ble and wifi mesh turn on together

MGDG, 2020-12-24:
## Environment

- Core:              ESP32-D0WD
- MDF version:  v1.0-beta1-129-g9212c07-dirty
- IDF version:    v3.3.4-371-g8ff6e2e
- wifi firmware version: 1425c12

## Problem Description

同时启用ble和wifi mesh后，程序出错，打印
Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled
依照提示在初始化wifi时将wifi ps设置为WIFI_PS_MIN_MODEM依旧存在该问题

按如下方式在启动mesh后再次调用esp_wifi_set_ps(WIFI_PS_MIN_MODEM)，然后再启动ble,问题貌似解决了,但是不确定是否存在隐患
...
MDF_ERROR_ASSERT(mwifi_start());
MDF_ERROR_ASSERT(esp_wifi_set_ps(WIFI_PS_MIN_MODEM));
...
start ble
...

## coredump

0x400929bf: invoke_abort at /home/lqh/esp/esp-mdf/esp-mdf-master-new-idf/esp-idf/components/esp32/panic.c:136
0x40092b41: abort at /home/lqh/esp/esp-mdf/esp-mdf-master-new-idf/esp-idf/components/esp32/panic.c:171
0x400fb22d: coex_init at ??:?
0x400d1cdb: esp_phy_rf_init at /home/lqh/esp/esp-mdf/esp-mdf-master-new-idf/esp-idf/components/esp32/phy_init.c:524
0x400d1d21: esp_phy_rf_init at /home/lqh/esp/esp-mdf/esp-mdf-master-new-idf/esp-idf/components/esp32/phy_init.c:524
0x400d1f30: esp_phy_load_cal_and_init at /home/lqh/esp/esp-mdf/esp-mdf-master-new-idf/esp-idf/components/esp32/phy_init.c:670
0x4016d310: esp_bt_controller_enable at /home/lqh/esp/esp-mdf/esp-mdf-master-new-idf/esp-idf/components/bt/bt.c:1529
0x4013b1f3: gatts_init at /home/lqh/esp/LING_MC/main/bluetooth/gatt_server/gatt_server.c:930
 (inlined by) gatts_task at /home/lqh/esp/LING_MC/main/bluetooth/gatt_server/gatt_server.c:1035

EspHuifeng, 2020-12-24:
@MGDG 这是一个已知问题，我们正在解决。后续会通过更新 idf 版本来修复该问题。

rikkapro0128, 2024-04-05:
> @MGDG 这是一个已知问题，我们正在解决。后续会通过更新 idf 版本来修复该问题。

hi there, so this problem is solve ?

tonyshuang, 2024-04-05:
您好，信件已收到。