espressif/esp-mdf #191: Possible Memory Leak in MQTT_example

peterNodens, 2021-01-08:
## Environment

- Development Kit:     ESP32-S2-Saola-1
- Core (if using chip or module): ESP32 WROOM S2
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): 1c76187
- Development Env:	VScode 
- Operating System:		Windows
- Power Supply:         USB

## Problem Description

Prior to this, I was working on a custom code which used the [mqtt_example](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/mqtt_example). I was experiencing some memory issues and so after removing my modifications, I went back to the start to test the base example. I believe I have found the issue in the mqtt_example, but I don't know what is causing it

I run a mosquitto MQTT broker on my PC, and subscribe to the mesh/+/toCloud topic via mqtt.fx

During normal running, monitoring the root node using `idf.py monitor ` , there is around 30kB of free heap available, according to the  `esp_get_free_heap_size()` in the `static void print_system_info_timercb(void *timer)`. I am attempting to test failure conditions to ensure that the system recovers as expected if an event such as the MQTT broker is shut down. 
When running the command `net stop mosquitto` from the command line to stop the MQTT broker, the ESP32 notifies on the monitor immediatley that the broker is down. 

However, if you simply turn off the PC WiFi connection, thus starving the broker of an internet connection, the ESP32 does not register this for some time. Furthermore, the free heap drops by around 1.5 - 2kB (when you have a single device connected) and more when more devices are connected. After around 40 seconds, the ESP32 will register the broker disconnect. When reconnecting the wifi of the PC, therefore restoring the broker internet connection, the ESP reconnects, but the memory assigned after the broker disconnect is never freed. Repeating the wifi disconnect multiple times eventually drops the free heap to the point where `W (552403) [mwifi, 1298]: <ESP_ERR_NO_MEM> Realloc size: 1456, new_ptr: 0x0, heap free: 2152` is posted. 

I cannot find what is taking the memory, or why it is not being freed once the connection is restored. When removing the `esp_mqtt_client_publish()` from mesh_mqtt_handle.c , the memory no longer drops, but of course, nothing is published. 

I am unsure if this is an issue related to mwifi, or if its specific to the mqtt_example, or even if its related to the IDF. 

### Expected Behavior
Upon restoring the MQTT broker internet connection, the memory assigned whilst the broker is off should be freed.  

### Actual Behavior
Upon restoring the MQTT broker internet connection, the memory assigned whilst the broker is off is not freed, and eventually the ESP32 will run out of memory. 

### Steps to repropduce
1. Run the mqtt_example.c on one or two ESP32S2 devices (two devices makes the problem more obvious) 
2. Set up an MQTT broker on a PC so that you can turn off the wifi connection
2. Monitor the root node using `idf.py monitor` - observe the free heap once the root task is started. In my case it was ~30kB
3. Once root is assigned and messages are being published, disconnect the broker's wifi connection
4. Observe the free heap printed in the monitor - in my case it drops around 2kB. 
5. Wait until the following is posted: 
```
D (90611) [mesh_mqtt, 216]: MQTT_EVENT_ERROR
E (90612) MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0
D (90613) [mesh_mqtt, 178]: MQTT_EVENT_DISCONNECTED
I (90617) [mqtt_examples, 246]: event_loop_cb, event: 24578
I (90624) [mqtt_examples, 313]: MQTT disconnected
I (90629) mesh: <MESH_NWK_ROOT_TODS_STATE>toDS:0
I (90634) [mwifi, 257]: State represents: 0
I (90638) [mqtt_examples, 246]: event_loop_cb, event: 11
W (90644) MQTT_CLIENT: Publish: Losing qos0 data when client not connected
```
6. Reconnect the broker wifi connection, and observe the free heap. In my case it decreases further, and is never recovered to the original ~30kB. 
7. Repeat steps 3-6 and eventually a no memory warning is posted. 

### Code to reproduce this issue

Run the [mqtt_example](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/mqtt_example). 

I have also tried placing
```
        heap_trace_start(HEAP_TRACE_LEAKS);
        ret = mesh_mqtt_write(src_addr, data, size, MESH_MQTT_DATA_JSON);
        heap_trace_stop();
        heap_trace_dump();
```
in the `root_write_task` but I don't think it gives much information with a stack level 4. I've included it as Log 2 
## Debug Logs
### Log 1
```
Executing action: monitor
Running idf_monitor in directory c:\users\peter\documents\nodenswork\esp32_mesh\mqtt_example
Executing "c:\Users\Peter\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe c:\Users\Peter\.espressif\esp-mdf\esp-idf\tools/idf_monitor.py -p COM5 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- c:\users\peter\documents\nodenswork\esp32_mesh\mqtt_example\build\mqtt_example.elf -m 'c:\Users\Peter\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe' 'c:\Users\Peter\.espressif\esp-mdf\esp-idf\tools\idf.py' '-p' 'COM5'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
[... 438 lines cut ...]
D (230733) [mesh_mqtt, 172]: MQTT_EVENT_CONNECTED
I (230733) [mqtt_examples, 246]: event_loop_cb, event: 24577
I (230733) [mqtt_examples, 299]: MQTT connect
I (230746) mesh: <MESH_NWK_ROOT_TODS_STATE>toDS:1
I (230747) [mwifi, 257]: State represents: 1
I (230750) [mqtt_examples, 246]: event_loop_cb, event: 11
D (230757) [mesh_mqtt, 184]: MQTT_EVENT_SUBSCRIBED, msg_id=4095
D (230762) [mesh_mqtt, 184]: MQTT_EVENT_SUBSCRIBED, msg_id=43005
D (230918) [mesh_mqtt, 184]: MQTT_EVENT_SUBSCRIBED, msg_id=28415
I (231959) [mqtt_examples, 193]: System information, channel: 11, layer: 1, self mac: 7c:df:a1:00:77:96, parent bssid: 4c:38:d8:24:b2:79, parent rssi: -44, node num: 1, free heap: 21912
I (231966) [mdf_mem, 122]: (839) <mwifi: 300> ptr: 0x3ffe3a70, size: 32
I (231973) [mdf_mem, 122]: (839) <mwifi: 305> ptr: 0x3ffe3af0, size: 176
I (231980) [mdf_mem, 122]: (1953) <mqtt_examples: 110> ptr: 0x3fff20c4, size: 1456
I (231989) [mdf_mem, 122]: (10459) <mwifi: 978> ptr: 0x3ffd63fc, size: 1456
I (231996) [mdf_mem, 122]: (231212) <mwifi: 1298> ptr: 0x3fffb004, size: 1456
I (232004) [mdf_mem, 129]: Memory record, num: 5, size: 4576
D (234205) [mqtt_examples, 161]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1007796", "parent":"4c38d824b279","layer":1}
D (237208) [mqtt_examples, 161]: Node send, size: 79, data: {"type":"heartbeat", "self": "7cdfa1007796", "parent":"4c38d824b279","layer":1}
```
### Log 2
[Log 2](https://github.com/peterNodens/Logs/blob/main/Log_2_issue.txt )

## Other items if possible

[SDK config ](https://github.com/peterNodens/Logs/blob/main/sdkconfig)

EspHuifeng, 2021-01-12:
Hi, @peterNodens.
Thank you for your report. I can reproduce this issue and I am debugging now.

peterNodens, 2021-01-12:
Hi @EspHuifeng . Glad to know you can replicate it and it is not just on my system. I look forward to any solutions you might have

EspHuifeng, 2021-01-13:
Hi @peterNodens, After several operations, will the free heap return to its original level over a period of time.

peterNodens, 2021-01-13:
Hi @EspHuifeng . Thanks for the reply and for looking into this. I have also seen that the memory recovers, however this seems to take a very random amount of time. Sometimes it recovers within a few minutes, other times it does not recover for over 30 minutes. 
Are you aware of what is consuming the memory, or what causes it to be freed? That way I can force it to free the memory quickly. I have tried tracing it but I have not had any success. 

The reason that this is a concern is that in my application, I am running another task on each node which is quite memory intensive, using ~14kB on each node. This is not a problem on the child nodes as there is plenty of memory available, but on the root node, this reduces the available heap to just 17kB before any child nodes are added. If the broker loses its internet connection for whatever reason, the available heap is dropping due to this issue. If two or three child nodes are running on the root (also consuming ~2kb each), and more than two broker disconnects happen within a short space of time, then the heap will drop to very low levels, possibly triggering errors and warnings.

EspHuifeng, 2021-01-14:
Hi, @peterNodens. 
This is the resource held by the socket, which is not immediately released when the connection is broken.

peterNodens, 2021-01-14:
Hi @EspHuifeng . Thanks, I thought that this might be the case as I could see the heap trace was going into the tcp. I was also looking at this: https://github.com/espressif/esp-idf/issues/2134#issuecomment-403336387 . Is that related? It seems like quite a lot of memory dissapears (~5kb) for around 30 minutes. Based on your responses, I am assuming that this is expected behaviour. 

Are there any settings we can change to at least help the situation? I've tried reducing the sockets and playing around with quite a few of the LWIP settings without much luck. For example, can we free the memory faster, or reduce the amount of memory the open socket is using? 

This issue is preventing us moving forward as we can't implement the mesh if it isn't resilient to more than 1 or 2 broker cut outs in a short period of time. Of course the mqtt_example can deal with a few cutouts, but in our implmentation that i mentioned above we get into a low heap situation which is quite dangerous. 

Thank you again.

EspHuifeng, 2021-01-15:
Hi, @peterNodens 
Maybe you can set is option `Component config -> LWIP -> Enable SO_LINGER processing`. This helps a little bit in shortening the time the socket takes to tie up resources. 
In addition, you can also configure the resource footprint of the wifi stack through this document to save memory.[How to configure parameters
](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#how-to-configure-parameters) Maybe there are other places that can save some memory.

peterNodens, 2021-01-15:
Hi @EspHuifeng thanks for these pointers. Since it seems there is no problem, I will close the issue. Thanks for your help!

closed 2021-01-15 by peterNodens