espressif/esp-mdf #192: A "train" topology MESH broadcast for GPIO manipulation.

Bytesforus, 2021-01-11:
I am currently assessing the future and possible use of a "train" MESH structure using up to 500-1000 ESP32`s for a specific application. A "train" topology currently does not seem to be available for the MESH as it stands, although promised.

Theoretical speeds mentioned for a ESP MESH seem to vary and practical speeds noted from several on-line projects also differ.

The application "train" MESH needs to broadcast down the line and manipulate the same GPIO (say 19) on all nodes (at speed).

Question 1: Is a "train" topology viable using WiFi on the ESP32 (or upcoming ICs) and will it be available in the future?
Question 2: What are the maximum (practical) speeds to be expected using the broadcast messaging system on a "train"?
Question 3: And what would be an optimal approach for reaching the goal of GPIO manipulation at the highest possible speeds?

The reason I ask is that before investing more time in further developing the ESP32 MESH I would like to know its (practical) capabilities for this type of application (I can not share details at this time, I am sorry).

If the capability is not there and will not be in the future one needs to start looking at other options. If you could clear this up?

will-emmerson, 2021-01-11:
Same question was asked here:
https://esp32.com/viewtopic.php?f=21&t=15025

What you're talking about isn't really a mesh and perhaps doesn't need to use mesh sdk / functions. You could maybe just do it manually with esp-now.

Bytesforus, 2021-01-11:
True, in some other topic on that same forum (I think) Espressif replied that they are working on the "train" topology (as a MESH).

Espressif considers it topology for a MESH, although I agree with you that strictly speaking it starts to resemble a new form.

ESP-NOW, possibly like MDF, is too heavy in the sense that it requires a minimum of 40 Bytes due to the frame format.

This means that theoretical throughput is reduced to about 1500 commands/sec at best. Not fast enough for the application.

Bytesforus, 2021-01-11:
Speed is set at 1Mbps default for the ESP-NOW, this does not necessarily mean it is the maximum but it possibly is a practical limit.

Bytesforus, 2021-01-11:
A lightweight protocol that utilizes a "train" where each Node relays a 1 Byte command to all Nodes within range might work. This means that the command Byte will duplicate (they sort of ripple and increase down the line) but it might increase overall speed as the furthest Node in terms of range will reach the following furthest Node thereby temporarily bypassing all intermediates.

One needs to implement a trigger on the (earliest) incoming command Byte in each Node so that Nodes only accept and relay commands once. A command Byte could contain a pre-set 7 bit “ID” and “command type” key ending with a 1 bit TRUE or FALSE statement to set a specific boolean. One could use different “command type” codes for specific program functions and also use them to configure and operate the “train” itself. The “ID” bits would be used to localize the "train" or to set and address groups.

This one Byte could employ some kind of basic security by the Node only accepting a limited number of pre-set “command types” from a specific “ID”. Both combined and encoded in the 7 bits to be received by the Node. Possibly adding time generated encryption to the Byte (or the 7 first bits) by using a timely shifted key from some hidden table inside the Node and RootNode.

It would mean that all Nodes are promiscuous but only accepting a specific command Byte one at a time. This time could be set.

A problem with successive ripples might be interference however after receiving the first 8 bits reception should essentially be blocked on a Node for time “x” but only for this specific command. This way other and new commands can follow immediately. Time “x” would be determined by “train” length, amount of Nodes, latency per Node and signal propagation speed (including the signal return path through space). This likely could lead to real signal interference. If so, a lightweight but chain-linked “train” protocol might be a better option. Also using a command Byte but addressing all intermediate Nodes in turn. I am not sure (at all) if WiFi allows for the suggested lightweight chain-less (or chained) “train” operation and if can be implemented in a practical way.

Bytesforus, 2021-01-11:
Real interference in a chain-less "train" could be solved by pre-fetching all the "furthest Nodes". These only would relay commands.

EspHuifeng, 2021-01-12:
Hi @Bytesforus @will-emmerson.
I think that you are talking about "chain topology", It is already supported by mesh(IDF v4.2 or above). The maximum layer can up to 1000. See [MESH_TOPO_CHAIN](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_mesh.html?highlight=mesh_topo_chain#_CPPv419esp_mesh_topology_t)
You can try it using MDF example. just set this configuration, https://github.com/espressif/esp-mdf/blob/1c76187faaf16d40a574fc7c0c98122e93dee746/components/mwifi/Kconfig#L67

Bytesforus, 2021-01-12:
Hello @ESPHuifens and thank you. Well yes, basicly, it depends on the throughput speed along the chain. I did not notice that the MESH offered that function as the reference is limited. It is not mentioned [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/mesh.html?highlight=mesh) for instance. Could you provide some insight into how the "chain" operates and what the (practical) throughput limits are in terms of speed? I would like to estimate the MESH performance at this time.

There are _possibly_ two ways to operate a train, chained (as now offered by Espressif) and unchained (this was my creative suggestion above in order to investigate a protocol for increased throughput speed but it might be problematic and unneccesary).

In case throughput speeds for the chained "train" topology are limited then I would like to develop an outlook on future options.

Bytesforus, 2021-01-12:
The 802.11n theoretical throughput speed stands at 30Mbps using UDP. The per "hop" latency stands at 10-30 milliseconds.

Air time for the bare 7 bits theoretically is 234 nanoseconds. 42000x faster than the lowest per hop latency currently achieved.

Therefore it would take a long time (considering its modern technology) to cross 1000 nodes, as it would take 10-31 seconds.

This means that currently the chained "train" topology is not viable for my application. Hence the suggestion for a new strategy.

I am interested to learn where bottlenecks are located in the "hop"? Could you also reflect on options for an unchained "train"?

Is it possible to adjust the current structure to a point where 7 bits of binary contain a 4 bit "ID" to address the "train"? Like so:
[image]

When I pull on a real-world rigid chain then the other end will move immediately. This is about what I am looking for this "train".

Bytesforus, 2021-01-13:
Considering the speeds required are not going to be reached anytime soon I close this

closed 2021-01-13 by Bytesforus