espressif/esp-mdf #194: Group Message with mdf mesh using esp_mesh_send()

Jigar3690, 2021-01-25:
Hello All,

I am trying to create a group within a mesh.
I am using ESP-IDF v3.3.4
I am using ESP-MDF v1.0

I am successfully using mdf mesh since a while now.
I have a fixed root talking to multiple leaf nodes.

I am trying to create a group within mesh, which I am actually able to create. My root and 2 nodes are now the member of my group.
After successfully creating the group I am now trying to send message to the group using the groupID.
I am using esp_mesh_send() function to send the data from root to the group members.
By using it I am continuously getting ESP_ERR_MESH_ARGUMENT error code in return.
I had make sure to set mesh_opt.type = MESH_OPT_SEND_GROUP and also flag is set to MESH_DATA_FROMDS.

I have a doubt that esp_mesh_send is the right function to use here or not.
I verified all documentation and it looks okay to me. There is nothing mentioned regarding whether the function can be used with mdf or not. 

Any help with this would be greatly appreciated.

EspHuifeng, 2021-02-18:
Hi @Jigar3690 
I tested the group function on [mwifi/router](https://github.com/espressif/esp-mdf/tree/master/examples/function_demo/mwifi/router) example and it works well. 
Could you test the group function on this example again?

Jigar3690, 2021-02-18:
I got that resolved. Thank you @EspHuifeng .

closed 2021-02-18 by Jigar3690