espressif/esp-mdf #198: [阿里云物联网平台SDK结合mconfig进行WIFI连接信息重配置的问题]

jayjayhust, 2021-02-16:
## Environment
- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM-32D]
- MDF version (``git rev-parse --short HEAD`` to get the commit id.): v1.0
- Development Env:		[other]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
在mdf\examples\aliyun_linkkit的工程下，结合mconfig的配网示例工程，想系统在WIFI信息正常的情况下是可以连通阿里云物联网平台的，如果需要更改WIFI设置，也可以通过按键长按进入配网模式，然后通过蓝牙进行WIFI连接信息的重配置，但是在进行配网处理的时候：
===========================================================
//获取网络设置（APP配网，比如乐鑫的ESPMesh小程序，此处逻辑参考mconfig的例程）
mdf_err_t get_network_config(const char *name, mwifi_config_t *mwifi_config, char custom_data[32])
{
    MDF_PARAM_CHECK(name);
    MDF_PARAM_CHECK(mwifi_config);
    MDF_PARAM_CHECK(custom_data);

    mconfig_data_t *mconfig_data        = NULL;
    mconfig_blufi_config_t blufi_config = {
        .tid = 1, /**< Type of device. Used to distinguish different products,
                       APP can display different icons according to this tid. */
        .company_id = MCOMMON_ESPRESSIF_ID, /**< Company Identifiers (https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers) */
    };

    strncpy(blufi_config.name, name, sizeof(blufi_config.name) - 1);
    MDF_LOGI("BLE name: %s", name);

    // Initialize Bluetooth network configuration
    **MDF_ERROR_ASSERT(mconfig_blufi_init(&blufi_config));**

    // Network configuration chain slave initialization for obtaining network configuration information from master.
    MDF_ERROR_ASSERT(mconfig_chain_slave_init());

    // Get Network configuration information from blufi or network configuration chain.
    //When blufi or network configuration chain complete, will send configuration information to config_queue.
    MDF_ERROR_ASSERT(mconfig_queue_read(&mconfig_data, portMAX_DELAY));

    // Deinitialize Bluetooth network configuration and Network configuration chain.
    MDF_ERROR_ASSERT(mconfig_chain_slave_deinit());
    MDF_ERROR_ASSERT(mconfig_blufi_deinit());

    memcpy(mwifi_config, &mconfig_data->config, sizeof(mwifi_config_t));
    memcpy(custom_data, &mconfig_data->custom, sizeof(mconfig_data->custom));

    // Switch to network configuration chain master mode to configure the network for other devices(slave), according to the white list.
    if (mconfig_data->whitelist_size > 0) {
        MDF_ERROR_ASSERT(mconfig_chain_master(mconfig_data, pdMS_TO_TICKS(60000)));
    }

    MDF_FREE(mconfig_data);

    return MDF_OK;
}
===========================================================
系统会卡死在“MDF_ERROR_ASSERT(mconfig_blufi_init(&blufi_config));”这个逻辑这里，然后断言失败系统重启。如果我禁掉系统初始化阶段的“MDF_ERROR_ASSERT(mwifi_start());”，系统不会卡死，但是系统阿里云的功能就没法正常运行了。

所以如何能够实现既不影响与阿里物联网平台的通信，又可以在系统进入配网模式后，能够重新通过蓝牙方式，使用mconfig组件实现WIFI信息的更新呢？

EspHuifeng, 2021-02-18:
Did you comment this line?
https://github.com/espressif/esp-mdf/blob/1c76187faaf16d40a574fc7c0c98122e93dee746/examples/aliyun_linkkit/mesh-with-aliyun/main/mesh_with_aliyun.c#L86

jayjayhust, 2021-02-19:
Hi Huifeng, thx for ur reply. No i haven't comment this line, need to do this?

jayjayhust, 2021-02-19:
> Did you comment this line?
> https://github.com/espressif/esp-mdf/blob/1c76187faaf16d40a574fc7c0c98122e93dee746/examples/aliyun_linkkit/mesh-with-aliyun/main/mesh_with_aliyun.c#L86

Hi Huifeng, thx for ur reply. No i haven't comment this line, need to do this?

closed 2021-02-20 by jayjayhust